zephyrproject-rtos/zephyr #52941: Zephyr assumes ARM M7 core has a cache

decsny, 2022-12-09:
**Describe the bug**
The issue is that the CPU_CORTEX_M7 Kconfig defined in arch/arm/core/aarch32/cortex_m/Kconfig forcefully selects the CPU_HAS_DCACHE and CPU_HAS_ICACHE Kconfigs, meaning that Zephyr is assuming that all M7 cores will have a cache. This is not true, because according to the M7 specification, the data and instruction caches are both optional

**Expected behavior**
I would expect that this Kconfig is set on a per soc basis.

**Impact**
The MPU code is actually what is affecting me. 

In arch/arm/core/aarch32/mpu/arm_mpu.c:

`#if defined(CONFIG_NOCACHE_MEMORY)
	/* Clean and invalidate data cache if it is enabled and
	 * that was not already done at boot
	 */
#if !defined(CONFIG_INIT_ARCH_HW_AT_BOOT)
	if (SCB->CCR & SCB_CCR_DC_Msk) {
		SCB_CleanInvalidateDCache();
	}
#endif
#endif /* CONFIG_NOCACHE_MEMORY */`

I believe this code causes a build error if Zephyr thinks the platform has a cache but the platform does not have definitions for these SCB cache functions, since it does not actually have a cache.

nashif, 2023-06-20:
@microbuilder  can you please look at this and give any useful feedback or close it if it is not a bug?

microbuilder, 2023-06-20:
I'm going to close this since I think having I/D cache on the M7 is the norm, and in the case of SoCs where there isn't a cache that can be indicated at the SoC level as an override. Feel free to open if you object @decsny or @carlocaione.

closed 2023-06-20 by microbuilder