zephyrproject-rtos/zephyr #52954: check_zephyr_package() only checks the first zephyr package rather than all the considered ones.

smrtos, 2022-12-11:
As I understand, the `check_zephyr_package()` is meant to find a better Zephyr package match.
But below code seems not exhausting all considered zephyr packages. They just return after running the first considered Zephyr package. Seems suspicious.

https://github.com/zephyrproject-rtos/zephyr/blob/87e63247c94e99d904f7d0c2f5dd84ddb309294f/share/zephyr-package/cmake/zephyr_package_search.cmake#L93-L94

https://github.com/zephyrproject-rtos/zephyr/blob/87e63247c94e99d904f7d0c2f5dd84ddb309294f/share/zephyr-package/cmake/zephyr_package_search.cmake#L96-L98

nordicjm, 2023-02-03:
How are you using zephyr? Can you give reproduction steps as to how you are setting this up?

tejlmand, 2023-03-02:
> Seems suspicious.

Thanks, we are doing some very special tricks to support Zephyr projects which may not be registered in the CMake package registry, so only good when someone flags things which may look suspicious.

But in this case, code is correct, which will be explained below.

> But below code seems not to be exhausting all considered zephyr packages. They just return after running the first considered Zephyr package. Seems suspicious.

Which in this case is absolutely correct, cause you missed the following three lines above:
https://github.com/zephyrproject-rtos/zephyr/blob/93c2e08711103f4415e3253ae6bf6a000d820619/share/zephyr-package/cmake/zephyr_package_search.cmake#L90

This code is only executed when `ZEPHYR_BASE` is already defined, meaning there is one and only one Zephyr to test and potentially use, and that's the Zephyr pointed to by `ZEPHYR_BASE`.
Likewise, `Zephyr_CONSIDERED_CONFIGS` is in this case a list of just one entry.

> Maybe we should check for that for each of the ZEPHYR_CANDIDATE before returning.

No, cause CMake's `find_package()` will do that check for us, and as mentioned above, when `ZEPHYR_BASE` is set, then that's the only Zephyr we should consider.

Closing, as this in not a bug.

closed 2023-03-02 by tejlmand