zephyrproject-rtos/zephyr #52970: samples/net/wifi example does not work with ESP32

SrivamsiMalladi, 2022-12-12:
I want to run Wi-Fi example on ESP32.  [#36414](https://github.com/zephyrproject-rtos/zephyr/issues/36414) says to use `samples/boards/esp32/wifi_station` but I could not find `wifi_station` in`samples/boards/esp32`. So I tired `samples/net/wifi` by followed the instructions in the [Wi-Fi Sample page](https://docs.zephyrproject.org/latest/samples/net/wifi/README.html) and I get the following linking error while building.

```
(.venv) iotdev@iotdev:~/zephyrproject/zephyr$ west build -p -b esp32 samples/net/wifi
-- west build: making build dir /home/iotdev/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/iotdev/zephyrproject/zephyr/samples/net/wifi
-- Found Python3: /home/iotdev/zephyrproject/.venv/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/iotdev/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/iotdev/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/home/iotdev/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/home/iotdev/zephyr-sdk-0.15.2)
-- Found Dtc: /home/iotdev/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/iotdev/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Found devicetree overlay: /home/iotdev/zephyrproject/zephyr/samples/net/wifi/boards/esp32.overlay
-- Generated zephyr.dts: /home/iotdev/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/iotdev/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/iotdev/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TEST_RANDOM_GENERATOR and/or look up
TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/iotdev/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/iotdev/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/iotdev/zephyrproject/zephyr/samples/net/wifi/prj.conf'
Merged configuration '/home/iotdev/zephyrproject/zephyr/samples/net/wifi/boards/esp32.conf'
Configuration saved to '/home/iotdev/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/iotdev/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
CMake Warning at ../../../CMakeLists.txt:824 (message):
  No SOURCES given to Zephyr library: drivers__ethernet

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/iotdev/zephyrproject/zephyr/build
-- west build: building application
[1/393] Preparing syscall dependency handling

[7/393] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/iotdev/zephyrproject/zephyr), build: zephyr-v3.2.0-2507-ga9481bdbe198
[376/393] Performing configure step for 'EspIdfBootloader'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
[... 109 lines cut ...]
[96/97] Generating binary image from built executable
esptool.py v3.3
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/iotdev/zephyrproject/zephyr/build/esp-idf/build/bootloader/bootloader.bin
[97/97] cd /home/iotdev/zephyrproject/zephyr/build/esp-idf/build/bootloader/esp-idf/esptool_py && /home/iotdev/zephyrproject/.venv/bin/python3.10 /home/iotdev/zephyrproject/modules/hal/espressif/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/iotdev/zephyrproject/zephyr/build/esp-idf/build/bootloader/bootloader.bin
Bootloader binary size 0x6520 bytes. 0xae0 bytes (10%) free.
[383/393] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/iotdev/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/wifi/libdrivers__wifi.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -L"/home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0"  -L/home/iotdev/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -T/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -L/home/iotdev/zephyrproject/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32 && cd /home/iotdev/zephyrproject/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
/home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: cannot find -lnet80211: No such file or directory
[... cut ...]  directory
/home/iotdev/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: cannot find -lrtc: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/iotdev/zephyrproject/zephyr/build
```

**Environment:**
 - OS: Ubuntu 20.04.1 LTS
 - Zephyr SDK: 0.15.2

sylvioalves, 2022-12-12:
Please, read, https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html. You might have missed `west blobs fetch hal_espressif`. Re-open if needed.

closed 2022-12-12 by sylvioalves

SrivamsiMalladi, 2022-12-13:
I am getting `SSLError` when trying the `west blobs fetch hal_espressif`.
```
(.venv) iotdev@iotdev:~/zephyrproject/zephyr$ west blobs fetch hal_espressif
Fetching blob hal_espressif: /home/iotdev/zephyrproject/modules/hal/espressif/zephyr/blobs/lib/esp32c3/libbtdm_app.a
Traceback (most recent call last):
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/connection.py", line 414, in connect
    self.sock = ssl_wrap_socket(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/util/ssl_.py", line 449, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/util/ssl_.py", line 493, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.10/ssl.py", line 513, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.10/ssl.py", line 1071, in _create
    self.do_handshake()
  File "/usr/lib/python3.10/ssl.py", line 1342, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:997)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/esp32c3-bt-lib/98dcc9591365b5ac486a9f0b474c36bf8c4ca97b/esp32c3/libbtdm_app.a (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:997)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/iotdev/zephyrproject/.venv/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/west/app/main.py", line 793, in main
    app.run(argv or sys.argv[1:])
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/west/app/main.py", line 109, in run
    self.run_command(argv)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/west/app/main.py", line 345, in run_command
    self.run_extension(args.command, argv)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/west/app/main.py", line 417, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/west/commands.py", line 135, in run
    self.do_run(args, unknown)
  File "/home/iotdev/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 140, in do_run
    subcmd(args)
  File "/home/iotdev/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 121, in fetch
    self.fetch_blob(blob['url'], blob['abspath'])
  File "/home/iotdev/zephyrproject/zephyr/scripts/west_commands/blobs.py", line 112, in fetch_blob
    inst.fetch(url, path)
  File "/home/iotdev/zephyrproject/zephyr/scripts/west_commands/fetchers/http.py", line 19, in fetch
    resp = requests.get(url)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/sessions.py", line 723, in send
    history = [resp for resp in gen]
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/sessions.py", line 723, in <listcomp>
    history = [resp for resp in gen]
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/sessions.py", line 266, in resolve_redirects
    resp = self.send(
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/home/iotdev/zephyrproject/.venv/lib/python3.10/site-packages/requests/adapters.py", line 563, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/esp32c3-bt-lib/98dcc9591365b5ac486a9f0b474c36bf8c4ca97b/esp32c3/libbtdm_app.a (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:997)')))
(.venv) iotdev@iotdev:~/zephyrproject/zephyr$ 
```

SrivamsiMalladi, 2022-12-13:
Also the `samples/net/wifi/src/wifi_test.c` is empty!!!!
And there is no `samples/boards/esp32/wifi_station` example application as suggested in some of the GitHub issues and archive posts. It would be really helpful if someone could point me to a Wi-Fi example.