zephyrproject-rtos/zephyr #52978: Consider Adding Realm to Zephyr

MarcusSorealheis, 2022-12-12:
## Origin

Realm
[GitHub Link](https://github.com/realm/realm-core)

## Purpose

Realm is a mobile database that runs directly in phones. Its permissive license (Apache 2.0), size, and development ergonomics make it an appealing persistence layer for other flavors of embedded Linux. The hope is that we can work together to more quickly solve problems by using the next evolution of persistence in resource constrained environments. In addition to being more flexible and natural in the object-oriented paradigm, it also natively supports better performance and security than SQLite out of the box.

## Mode of integration

Open to advice here.

## Maintainership

@marcussorealheis (I will quickly replace myself with other, smarter folks)

## Pull Request

Pull request (if any) with the actual implementation of the integration, be it
in the main tree or as a module (pointing to your own fork for now). Make sure
the PR is correctly labeled as "DNM"

## Description

Long description that will help reviewers discuss suitability of the
component to solve the problem at hand (there may be a better options
available.)

What is its primary functionality?
Realm is a lightweight and natural database

What problem are you trying to solve? (e.g., a state store is
required to maintain ...)
Development productivity and state storage. Solving these two problems will unlock all sorts of possibilities.

Why is this the right component to solve it?
Realm is small, easier to use than SQLite, and has a very liberal license

## Dependencies

What other components does this package depend on?

1. build-essential
2. libcurl4-openssl-dev 
3. libuv1-dev
4. libprocps-dev
5. libssl-dev
6. zlib1g-dev
7. cmake

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?

Hopefully the latter. 

## Revision

Version or SHA you would like to integrate initially

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``Apache-2.0``

nordicjm, 2022-12-13:
> How small? A few kilobytes. Realm with no data is mostly a schema and a file system akin to SQLite.

I'm not talking about the size of the database, I'm talking about the size of the engine itself. Zephyr micros have 16KB - 256KB of RAM, you can't compare an embedded micro like this with an SoC that you'd get in a phone which this database is designed for, which is going to have no less than 1GB or 2GB of RAM. If this database engine (I'm pulling a number out of thin air here) needs e.g. 4MB of RAM to do things with a small database, it does not make sense to port it because it would basically be useless.

MarcusSorealheis, 2022-12-14:
The thin air number doesn't seem too off for many cases. I thought the range of Zephyr devices was broader than 16KB-256KB. I overestimated Moore's law. 

In fact, in my automotive and IoT days, people were exploring Zephyr for boards that were more mini than micro I suppose? It looks like some of the [NXP ARM boards](https://docs.zephyrproject.org/3.2.0/boards/arm/mimx8mm_evk/doc/index.html) are still in the Zephyr mix. Boards in this family are good for a broad swathe of use cases. 

I will lean on your recommendation because I'm ultimately here to learn and contribute. I also intend to bring a new wave of supporters because I believe in open source. So, whatever advice or critique, I am all ears. I can further refine the scope of my issue if that would be helpful.

nashif, 2024-02-12:
outdated, closing. Please open new issue with updated information if this is still needed.

closed 2024-02-12 by nashif