zephyrproject-rtos/zephyr #52981: LOG_MODE_MINIMAL do not work with USB_CDC for nRF52840

DuMaM, 2022-12-12:
**Describe the bug**

I was trying to enable a log in console.
To do so, I played around with `shell_module` example, and I was trying to add another console driver to this example.
When I pick `LOG_DEFFERED=y` it works perfectly.
When I pick `LOG_MINIMAL or LOG_MODE_IMMEDIATE` then nothing works. 

**To Reproduce**
Steps to reproduce the behavior:
1. Take a nRF52840 board
2. Change LOG_MODE_DEFFERED to LOG_MODE_MINIMAL and add `zephyr,console = &cdc_acm_uart0;` in usb.overlay file in `shell_subsystem` sample.
3. build sample with your changes together with prj_minimal.conf with usb_overlay.proj 
4. See error

**Expected behavior**
Shell and console is working on usb port.

**Impact**
Huge for me. I need a solution which gives me fast output.
For others, I guess it will be not a big problem.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain NRF SDK 2.2.0,
 - Commit SHA commit c0689b16ff127d1b71a4cc20310d476e1807e26e (3.2.99)

DuMaM, 2022-12-12:
Here is repo with my changes
https://github.com/DuMaM/nrf_submodule_shell_error/commits/master

desowin, 2022-12-13:
I don't think going with LOG_MODE_MINIMAL will give you the desired fast output due to the very nature how USB works (host initiates every single transaction). The biggest limiting factor currently is that when CDC ACM is used as console, then every single character output to console tends to get transmitted in its own transfer. That is, on host you get 1 byte per URB which is seriously limiting the throughput.

The console uart driver writes the characters with `uart_poll_out`:
https://github.com/zephyrproject-rtos/zephyr/blob/85a7d0cc467384be33ce7a6d80f35cefce1d6d1b/drivers/console/uart_console.c#L77-L95
which calls `cdc_acm_poll_out`:
https://github.com/zephyrproject-rtos/zephyr/blob/85a7d0cc467384be33ce7a6d80f35cefce1d6d1b/subsys/usb/device/class/cdc_acm.c#L988-L1020

`USB_WORKQUEUE_PRIORITY` Kconfig help mentions:
```
  By default, USB work queue priority is the lowest cooperative
  priority. This means that any work handler, once started, won't
  be preempted by any other thread until finished.
```

Therefore, if the console output happens in lower priority (i.e. any preemptive priority) then you'll essentially get one character at a time. You can get more characters at a time (which allows greater throughput, which is essentially what @DuMaM seems to want) by changing USB_WORKQUEUE_PRIORITY to lower priority (e.g. to 1; lowest cooperative priority is higher than any preemptive priority) but that might bring unforeseen issues.

@pfalcon Any idea how to handle it? I think the USB workqueue being cooperative is generally good, however it clearly messes up with the possibility to fill up console buffer.

desowin, 2022-12-13:
@DuMaM Could you please check if #53003 gives you enough speedup when used with deferred logging?

desowin, 2022-12-14:
Minimal logging is working just fine with CDC ACM console when I disable CDC ACM logging. I think the only way around that is to either remove the CDC ACM logging completely or to fail build if CDC ACM is used as console and CDC ACM log level is not Off (done in #53032).

PR #53032 (2022-12-15): usb: device: cdc_acm: Prevent recursive logging loop
Allow enabling CDC ACM logging only if CDC ACM is not used as logging backend. This prevents endless recursive logging loop, especially visible when minimal footprint logging is enabled.

Fixes: #52981
commit 21975231 (2022-12-15): usb: device: cdc_acm: Prevent recursive logging loop

Allow enabling CDC ACM logging only if CDC ACM is not used as logging
backend. This prevents endless recursive logging loop, especially
visible when minimal footprint logging is enabled.

Fixes: #52981