zephyrproject-rtos/zephyr #52983: Bluetooth: Audio: BT_CODEC_LC3_CONFIG_DATA fails to compile with _frame_blocks_per_sdu > 1

Chris-StJohn-Bose, 2022-12-12:
When attempting to create a LC3 config with more than one frame block per SDU the compiler hits an undefined symbol `_frames_per_sdu`

```
#define BT_CODEC_LC3_CONFIG_DATA(_freq, _duration, _loc, _len, _frame_blocks_per_sdu) \
{ \
	BT_CODEC_DATA(BT_CODEC_CONFIG_LC3_FREQ, _freq), \
	BT_CODEC_DATA(BT_CODEC_CONFIG_LC3_DURATION, _duration), \
	BT_CODEC_DATA(BT_CODEC_CONFIG_LC3_CHAN_ALLOC, \
		      ((_loc) & 0xFFU), \
		      (((_loc) >> 8) & 0xFFU), \
		      (((_loc) >> 16) & 0xFFU), \
		      (((_loc) >> 24) & 0xFFU)), \
	BT_CODEC_DATA(BT_CODEC_CONFIG_LC3_FRAME_LEN, \
		      ((_len) & 0xFFU), \
		      (((_len) >> 8) & 0xFFU)) \
	COND_CODE_1(_frame_blocks_per_sdu, (), \
		    (, BT_CODEC_DATA(BT_CODEC_CONFIG_LC3_FRAME_BLKS_PER_SDU, \
				     _frames_per_sdu))) \
}
```

PR to follow....

Chris-StJohn-Bose, 2022-12-12:
#52986

kruithofa, 2023-02-02:
@Thalley @Chris-StJohn-Bose can you confirm that #52986 fixes this issue?

Thalley, 2023-02-02:
> @Thalley @Chris-StJohn-Bose can you confirm that #52986 fixes this issue?

It does, yes. Closing the issue :)

closed 2023-02-02 by Thalley