zephyrproject-rtos/zephyr #52987: Support Uptane as a way to update Zephyr software

JustinCappos, 2022-12-12:
## Introduction

Right now, Zephyr updates seem to mostly be using the SWupdate tool.  We'd like to include an alternative tool, Uptane, which provides a much stronger set of security properties that help it resist even nation-state actors. 

### Problem description

SWupdate does not provide revocation, delegations, roles, or similar mechanisms to account for situations where the signing key and/or repository are compromised.  

### Proposed change

I'd like to help the Uptane project, specifically, the [Aktualizr](https://github.com/advancedtelematic/aktualizr) implementation which is included in AGL, etc. be an option for interested users.

## Detailed RFC

For details about the security model, etc, please see the [Uptane website](https://uptane.github.io/).  Essentially, right now, if a key or server is compromised, there is a serious risk to all Zephyr users.  Adopting a different architecture will mitigate much of this risk.

### Proposed change (Detailed)

I'll have to have someone from our team take a look at how to package this all up and get things smoothly into Zephyr.  I'm curious to talk first to explore if this is of interest before working to do this effort.

### Dependencies

I'm not sure.  Will need to consult with others.  Just starting the process here first.   Overall the codebase is used and maintained by Toradex, which runs Zephyr on some of their boards.  

### Concerns and Unresolved Questions

Need to have further discussions with the community about any concerns from their end.

## Alternatives

I am also speaking with the folks at SWupdate about possibly adding the security model from Uptane (or similar) to their tool.  I would be happy to see that effort progress also or possibly instead of this.

beriberikix, 2022-12-16:
@kestewart asked for my .02, so sharing the wealth :) Generally I think it's a great thing to provide more options for [device management](https://docs.zephyrproject.org/latest/services/device_mgmt/index.html) & [OTA](https://docs.zephyrproject.org/latest/services/device_mgmt/ota.html). So thoughts that come to mind:

- What aspects can be generalized/re-used by other subsystem like [DFU](https://docs.zephyrproject.org/latest/services/device_mgmt/dfu.html#device-firmware-upgrade)
- What are the target class of the devices? The typical Zephyr MCU device is rather constrained and I *think* the aktualizr client typically runs on a Linux-class device.
- Zephyr's C++ support is functional but limited. C modules are generally preferred in tree.

FWIW, at Golioth we use our own mechanism for OTA and IETF SUIT but I would be excited to see any improvements to this part of Zephyr!

ceolin, 2022-12-16:
Hi, I agree with @beriberikix. aktualizr client will not just fit on Zephyr, unless you are intending to pull an specific part. I'm also interested to understand what are the intended targets and how it relates with other components like boot loader.
From the C++ thing, don't take for granted that any C++ feature is available and take in mind we support multiple toolchains.

That's said, it will be fantastic to have it. I'll take a closer look in the client to give more feedback.

henrikbrixandersen, 2023-01-03:
You mention that Zephyr OTA is currently mostly done using [SWupdate](https://swupdate.org/). I am not aware of a Zephyr port of SWupdate, could you please post a reference to this?

jameswalmsley, 2026-01-07:
The original project proposed by the oriiginal question has been archived.
I'm sure we can close this issue.

https://github.com/advancedtelematic/aktualizr

closed 2026-01-07 by henrikbrixandersen