zephyrproject-rtos/zephyr #52996: config parser

albertofloyd, 2022-12-13:
**Describe the bug**
When passing multiple filenames separated with either spaces or semicolons to OVERLAY_CONFIG, only 1 gets merged.

**To Reproduce**
**Add following files into samples/helloworld**
    extra1.conf
`    CONFIG_PMW=y
`
    extra2.conf
`    CONFIG_SENSOR=y
`

Steps to reproduce the behavior:
1. cd samples/helloworld
2. rm -rf build && west build -c -p always -b mec172xevb_assy6906 -- -DOVERLAY_CONFIG=extra1.conf;extra2.conf
3. grep -r CONFIG_PWM .
4. grep -r CONFIG_SENSOR .

CONFIG_PWM is enabled, but CONFIG_SENSOR is not

**Expected behavior**
OVERLAY_CONFIG should allow multiple filenames can be separated with either spaces or semicolons per documentation.

> OVERLAY_CONFIG: Additional Kconfig configuration fragment files. Multiple filenames can be separated with either spaces or semicolons. This can be useful in order to leave CONF_FILE at its default value, but “mix in” some additional configuration options.

https://docs.zephyrproject.org/latest/develop/application/index.html

**Logs and console output**
Parsing /home/jamezaar/ecwork/zephyr_albertofloyd/Kconfig
Loaded configuration '/home/jamezaar/ecwork/zephyr_albertofloyd/boards/arm/mec172xevb_assy6906/mec172xevb_assy6906_defconfig'
Merged configuration '/home/jamezaar/ecwork/zephyr_albertofloyd/samples/hello_world/prj.conf'
Merged configuration '/home/jamezaar/ecwork/zephyr_albertofloyd/samples/hello_world/**extra1.conf**'
Configuration saved to '/home/jamezaar/ecwork/zephyr_albertofloyd/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/jamezaar/ecwork/zephyr_albertofloyd/samples/hello_world/build/zephyr/include/generated/autoconf.h

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK, ...)

nordicjm, 2022-12-13:
Not a bug, you are not encapsulating the command in quotes, so when you do `rm -rf build && west build -c -p always -b mec172xevb_assy6906 -- -DOVERLAY_CONFIG=extra1.conf;extra2.conf` what you actually do is run 2 commands, the first: `rm -rf build && west build -c -p always -b mec172xevb_assy6906 -- -DOVERLAY_CONFIG=extra1.conf` and the second `extra2.conf`. This will include both: `rm -rf build && west build -c -p always -b mec172xevb_assy6906 -- "-DOVERLAY_CONFIG=extra1.conf;extra2.conf"`. Closing

closed 2022-12-13 by nordicjm

albertofloyd, 2022-12-14:
Thanks @nordicjm