zephyrproject-rtos/zephyr #52998: tests: drivers: can: Build failure with sysroot path not quoted on Windows

ccli8, 2022-12-13:
**Describe the bug**
Try to manually build CAN test samples `tests/drivers/can/api` and meet compile error as below. Per my preliminary inspect, the `sysroot` path is not quoted, and space characters in the path cause such error, though, it is quoted elsewhere.

**To Reproduce**
To reproduce the behavior, just compile by:
```
west -v build -b frdm_k64f
```

**Expected behavior**
Compile error with log below

**Impact**
N/A

**Logs and console output**
```
[161/182] cmd.exe /C "cd /D C:\CCLi8\IoT\zephyr\zephyrproject\build\zephyr && C:\Python38\python.exe C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/scripts/build/gen_app_partitions.py -e C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/zephyr_pre0.elf -o C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated/app_smem_aligned.ld"
[162/182] cmd.exe /C "cd /D C:\CCLi8\IoT\zephyr\zephyrproject\build\zephyr && C:\Python38\python.exe C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/scripts/build/gen_kobject_list.py --kernel C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/zephyr_pre0.elf --gperf-output kobject_prebuilt_hash.gperf --include C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/misc/generated/struct_tags.json "
[163/182] cmd.exe /C "cd /D C:\CCLi8\IoT\zephyr\zephyrproject\build\zephyr && C:\ProgramData\chocolatey\bin\gperf.exe --output-file kobject_prebuilt_hash_preprocessed.c --multiple-iterations 10 kobject_prebuilt_hash.gperf"
[164/182] cmd.exe /C "cd /D C:\CCLi8\IoT\zephyr\zephyrproject\build\zephyr && C:\Python38\python.exe C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/scripts/build/process_gperf.py -i kobject_prebuilt_hash_preprocessed.c -o kobject_prebuilt_hash.c -p "struct z_object" "
[165/182] C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR19DD~1.EXE -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE*1024 -DCPU_MK64FN1M0VLL12 -DKERNEL -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include -IC:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/ztest/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/atmel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/modules/hal_nxp/. -isystem C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include" -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include-fixed" -fno-strict-aliasing -Os -imacros C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/arm-none-eabi -imacros C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wformat -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/tests/drivers/can/api=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -fno-function-sections -fno-data-sections -MD -MT zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj -MF zephyr\CMakeFiles\kobj_prebuilt_hash_output_lib.dir\kobject_prebuilt_hash.c.obj.d -o zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj -c C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/kobject_prebuilt_hash.c
FAILED: zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj 
C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR19DD~1.EXE -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE*1024 -DCPU_MK64FN1M0VLL12 -DKERNEL -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include -IC:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/ztest/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/atmel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/modules/hal_nxp/. -isystem C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include" -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include-fixed" -fno-strict-aliasing -Os -imacros C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/arm-none-eabi -imacros C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wformat -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/tests/drivers/can/api=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -fno-function-sections -fno-data-sections -MD -MT zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj -MF zephyr\CMakeFiles\kobj_prebuilt_hash_output_lib.dir\kobject_prebuilt_hash.c.obj.d -o zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj -c C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/kobject_prebuilt_hash.c
AR19DD~1.EXE: error: Files: No such file or directory
AR19DD~1.EXE: error: (x86)/GNU: No such file or directory
AR19DD~1.EXE: error: Arm: No such file or directory
AR19DD~1.EXE: error: Embedded: No such file or directory
AR19DD~1.EXE: error: Toolchain/10: No such file or directory
AR19DD~1.EXE: error: 2021.07/arm-none-eabi: No such file or directory
ninja: build stopped: subcommand failed.
```

Note `sysroot` path is not quoted:
```
--sysroot=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/arm-none-eabi
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain GNU Arm Embedded Toolchain 10.3-2021.07
 - Commit e2742af6ed1118e1e2dd2b0ffe89ce5c210ab22d
 - Python: 3.8.10
 - CMake: 3.24.3
 - Ninja: 1.11.1
 - west: 0.14.0

**Additional context**
N/A

carlescufi, 2022-12-13:
We don't support paths with spaces, IIRC.

ccli8, 2022-12-14:
> We don't support paths with spaces, IIRC.

It isn't consistent. Actually, there have been many `sysroot` parameters specified in front, and they are all quoted like `"--sysroot=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/arm-none-eabi"`, except the last one shown above which isn't quoted and causes this error. Additionally, on Windows, the Arm GCC toolchain path `C:/Program Files (x86)/GNU Arm Embedded Toolchain` is default and very common, though, it has space characters.

```
[7/182] cmd.exe /C "cd /D C:\CCLi8\IoT\zephyr\zephyrproject\build\zephyr && C:\Python38\python.exe C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/scripts/build/gen_offset_header.py -i C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj -o C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated/offsets.h"
[8/182] C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR19DD~1.EXE -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE*1024 -DCPU_MK64FN1M0VLL12 -DKERNEL -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/kernel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/arch/arm/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include -IC:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/ztest/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/atmel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/modules/hal_nxp/. -isystem C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include" -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include-fixed" -fno-strict-aliasing -Os -imacros C:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee "--sysroot=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/arm-none-eabi" -imacros C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/tests/drivers/can/api=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj -MF zephyr\CMakeFiles\zephyr.dir\lib\os\dec.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj -c C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/os/dec.c
[9/182] C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR19DD~1.EXE -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE*1024 -DCPU_MK64FN1M0VLL12 -DKERNEL -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/kernel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/arch/arm/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/include -IC:/CCLi8/IoT/zephyr/zephyrproject/build/zephyr/include/generated -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/include -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/subsys/testsuite/ztest/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/atmel/include -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -IC:/CCLi8/IoT/zephyr/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -IC:/CCLi8/IoT/zephyr/zephyrproject/zephyr/modules/hal_nxp/. -isystem C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/libc/minimal/include -isystem "c:/program files (x86)/gnu arm embedded toolchain/10 2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include" -isystem "c:/program files (x86)/gnu arm embedded toolchai [... cut ...] hyrproject/zephyr/tests/drivers/can/api=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/CCLi8/IoT/zephyr/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj -MF zephyr\CMakeFiles\zephyr.dir\lib\os\thread_entry.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj -c C:/CCLi8/IoT/zephyr/zephyrproject/zephyr/lib/os/thread_entry.c
```

closed 2023-02-28 by github-actions[bot]

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".