zephyrproject-rtos/zephyr #53006: Hawbkit with b_l4s5i_iot01a - wifi_eswifi: Cannot allocate rx packet

jeronimoagullo, 2022-12-13:
**Describe the bug**
The image download is not working correctly for the b_l4s5i_iot01a board due to `rx packet allocation error` with the `esWifi` module. The hawkbit code is correct since I have tested successfully the same code for esp32 board.

**Environment**
**Target board:** b_l4s5i_iot01a
**Testing app:** samples/subsys/mgmt/hawkbit
**Zephyr version:** 3.2.99
**Zephyr SDK:** 0.15.0

**prj.conf**
```
#Turn on the Hawkbit
CONFIG_HAWKBIT=y

#Main Stack Size
CONFIG_MAIN_STACK_SIZE=4096

#Minimal Heap mem pool size for the Hawkbit working
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096

CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n

#Enable DHCPv4 server
CONFIG_NET_DHCPV4=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

#Enable DNS Resolver
CONFIG_DNS_RESOLVER=y
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="8.8.8.8"
CONFIG_DNS_SERVER2="192.168.1.1"

#Optional if you would like test on the your server
CONFIG_SHELL=y
CONFIG_HAWKBIT_SHELL=y
CONFIG_KERNEL_SHELL=y
CONFIG_SHELL_STACK_SIZE=4096

#Hawkbit polling mode
CONFIG_HAWKBIT_POLLING=y

#Config Hawkbit Server
CONFIG_HAWKBIT_SERVER="192.168.1.100"

#Debug helpers
CONFIG_LOG=y
CONFIG_HAWKBIT_LOG_LEVEL_DBG=y

#General Config
CONFIG_NEWLIB_LIBC=y
CONFIG_LOG_BUFFER_SIZE=4096

#Generate HEX output
CONFIG_BUILD_OUTPUT_HEX=y

# for b_l4s5i_iot01a  board
CONFIG_NET_SOCKETS_OFFLOAD=y
CONFIG_WIFI=y
CONFIG_WIFI_ESWIFI=y
CONFIG_NET_L2_WIFI_MGMT=y

# mcuboot configuration
# Ensure an MCUboot-compatible binary is generated.
CONFIG_BOOTLOADER_MCUBOOT=y
# The key
CONFIG_MCUBOOT_GENERATE_UNSIGNED_IMAGE=n
CONFIG_MCUBOOT_SIGNATURE_KEY_FILE="mykey.pem"
[... 158 lines cut ...]
[00:00:19.574,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:19.726,000] <dbg> hawkbit: response_cb: Download percentage: 43%
[00:00:19.811,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.013,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.019,000] <dbg> hawkbit: response_cb: Download percentage: 44%
[00:00:20.183,000] <dbg> hawkbit: response_cb: Download percentage: 45%
[00:00:20.268,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.497,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.509,000] <dbg> hawkbit: response_cb: Download percentage: 46%
[00:00:20.711,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.920,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:20.965,000] <dbg> hawkbit: response_cb: Download percentage: 47%
[00:00:21.196,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:21.202,000] <dbg> hawkbit: response_cb: Download percentage: 48%
[00:00:21.410,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:21.560,000] <dbg> hawkbit: response_cb: Download percentage: 49%
[00:00:21.638,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:46.189,000] <err> hawkbit: Firmware - flash validation has failed
[00:00:46.228,000] <inf> hawkbit: Update failed
```

**Additional context**
I don't know if the error arises due to the eswifi module (dropping messages) or due to how the new messages are handled in Zephyr. 

Thank you in advance

jeronimoagullo, 2022-12-13:
I would like to share a little more verbose log. The error seems to come from how the network packages are handled in `net_pkt.c` file.

```
[00:00:23.557,000] <dbg> net_http: on_body: (0x20008078): Processed 33521 length 572
[00:00:23.557,000] <dbg> net_http: http_wait_data: (0x20008078): Calling callback for partitioned 2000 len data
[00:00:23.557,000] <inf> Sngular_hawkbit_rollout: HAWKBIT_DOWNLOAD
[00:00:23.557,000] <dbg> Sngular_hawkbit_rollout: response_cb: hb_context.dl.http_content_size: 198900
[00:00:23.603,000] <dbg> Sngular_hawkbit_rollout: response_cb: Download percentage: 16%
[00:00:23.687,000] <err> net_pkt: Data buffer (1460) allocation failed (eswifi_off_read_work:142)
[00:00:23.687,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:23.687,000] <dbg> net_http: on_body: (0x20008078): Processed 34409 length 888
[00:00:23.882,000] <err> net_pkt: Data buffer (1460) allocation failed (eswifi_off_read_work:142)
[00:00:23.882,000] <err> wifi_eswifi: Cannot allocate rx packet
[00:00:23.882,000] <dbg> net_http: on_body: (0x20008078): Processed 35521 length 1112
[00:00:23.882,000] <dbg> net_http: http_wait_data: (0x20008078): Calling callback for partitioned 2000 len data
[00:00:23.882,000] <inf> Sngular_hawkbit_rollout: HAWKBIT_DOWNLOAD
[00:00:23.882,000] <dbg> Sngular_hawkbit_rollout: response_cb: hb_context.dl.http_content_size: 198900
[00:00:23.906,000] <dbg> Sngular_hawkbit_rollout: response_cb: Download percentage: 17%
[00:00:23.934,000] <dbg> net_http: on_body: (0x20008078): Processed 35821 length 300
[00:00:24.045,000] <dbg> net_http: on_body: (0x20008078): Processed 37281 length 1460
[00:00:24.157,000] <dbg> net_http: on_body: (0x20008078): Processed 37289 length 8
[00:00:24.241,000] <err> net_pkt: Data buffer (1460) allocation failed (eswifi_off_read_work:142)
[00:00:24.241,000] <err> wifi_eswifi: Cannot allocate rx packet
```

Besides, after some tests, the downloaded data is roughly the middle of the total binary size (around 50%) even if it is a large file (200k hawkbit app) or a small one (14k hello world app).

erwango, 2022-12-16:
I suspect a RX stack size that has to be increased somewhere, but @loicpoulain would tell better which one.

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".

closed 2023-05-17 by kartben

LeoBRIANDSmile, 2024-07-30:
> Closing with the correct reason as this was improperly marked as "completed".

Hi, it's a bit late, but I think I found a solution @kartben

erwango, 2024-07-30:
> > Closing with the correct reason as this was improperly marked as "completed".
> 
> Hi, it's a bit late, but I think I found a solution @kartben

It's never too late to fix issues. Don't hesitate to contribute your fix and provide reference to this issue.