zephyrproject-rtos/zephyr #53007: Bluetooth: Notification callback is called with incorrect connection reference

ahmedmoheb-nordic, 2022-12-13:
**Describe the bug**
Central connects to 2 peripherals and subscribes for notifications.
Notification callback is called with incorrect connection reference.

**To Reproduce**
Steps to reproduce the behavior:
1. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Expected behavior**
- Notifications are sent in both directions between the central and connected peripherals
- For each received notification packet, correct connection reference is reported by the notification callback function

**Logs and console output**
Please check the attached log
[log.log](https://github.com/zephyrproject-rtos/zephyr/files/10217517/log.log)

1. At line 548, Central received a notification with incorrect connection reference and incorrect destination address.
2. At line 563, the same notification packet was received again with the correct connection reference and correct destination address

[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain : Zephyr SDK 0.15.1
 - Commit SHA 7c7dee105f42bafece900df44202689efc6d30c9

jori-nordic, 2022-12-14:
After sprinkling some logs, it seems it is the controller that messes up the connection handle (last packet dumped in the log).
Maybe a race condition in the ctlr's HCI layer?

```
d_00: @00:00:09.994131  [00:00:09.994,110] <err> bt_ctlr_hci: hci_acl_encode: handle: 1
d_00: @00:00:09.994131  [00:00:09.994,110] <err> bt_ctlr_hci: hci_acl_encode: PDU data
d_00: @00:00:09.994131                                        df 00 04 00 1b 12 00 43  6f 75 6e 74 65 72 3a 31 |.......C ounter:1
d_00: @00:00:09.994131                                        38 34 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |84...... ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.994131                                        00 00 00                                         |...              
d_00: @00:00:09.994131  [00:00:09.994,110] <dbg> bt_ctlr_hci_driver: recv_thread: Packet in: type:3 len:231
d_00: @00:00:09.994131  [00:00:09.994,110] <dbg> bt_l2cap: bt_l2cap_recv: Packet for CID 4 len 223
d_00: @00:00:09.994131  [00:00:09.994,110] <dbg> bt_l2cap: l2cap_chan_recv: chan 0x80de7d0 len 223
d_00: @00:00:09.994131  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 1 data Counter:184 length 220 cnt 184
d_00: @00:00:09.994131  [00:00:09.994,110] <dbg> bt_ctlr_hci_driver: recv_thread: blocking
d_02: @00:00:09.994449  [00:00:09.994,445] <err> bt_ctlr_hci: hci_acl_encode: handle: 0
d_02: @00:00:09.994449  [00:00:09.994,445] <err> bt_ctlr_hci: hci_acl_encode: PDU data
d_02: @00:00:09.994449                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_02: @00:00:09.994449                                        00 00 00 00 00 00 00 00  00 00 00                |........ ...     
d_00: @00:00:09.995567  [00:00:09.995,544] <dbg> bt_ctlr_hci_driver: prio_recv_thread: sem taken
d_00: @00:00:09.995567  [00:00:09.995,544] <dbg> bt_ctlr_hci_driver: prio_recv_thread: RX node enqueue
d_00: @00:00:09.995567  [00:00:09.995,544] <dbg> bt_ctlr_hci_driver: prio_recv_thread: sem take...
d_00: @00:00:09.995567  [00:00:09.995,544] <dbg> bt_ctlr_hci_driver: recv_thread: unblocked
d_00: @00:00:09.995567  [00:00:09.995,544] <err> bt_ctlr_hci: hci_acl_encode: handle: 1
d_00: @00:00:09.995567  [00:00:09.995,544] <err> bt_ctlr_hci: hci_acl_encode: PDU data
d_00: @00:00:09.995567                                        df 00 04 00 1b 12 00 43  6f 75 6e 74 65 72 3a 31 |.......C ounter:1
d_00: @00:00:09.995567                                        38 35 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |85...... ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
d_00: @00:00:09.995567                                        00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[... cut ...]
d_00: @00:00:09.998751  [00:00:09.998,748] <dbg> bt_l2cap: bt_l2cap_recv: Packet for CID 4 len 223
d_00: @00:00:09.998751  [00:00:09.998,748] <dbg> bt_l2cap: l2cap_chan_recv: chan 0x80de8b8 len 223
d_00: @00:00:09.998751  [Central] : [NOTIFICATION] addr CB:04:A6:15:1F:50 (random) conn 0 data Counter:187 length 220 cnt 187
d_00: @00:00:09.998751  [Central] notify_func:185 : expected counter : 0 , received counter : 187
d_00: @00:00:09.998751 ERROR: (WEST_TOPDIR/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/central/src/main.c:187): [Central] wrong counter
```

cvinayak, 2022-12-14:
I am not sure if the test failure is due to incorrect connection handle/reference.
1. I got supervision timeout on first connection, either of the two below should get the test running with 2 active peripheral connections from the central.
Please try with this workaround: https://github.com/zephyrproject-rtos/zephyr/pull/51648/files
The open issue: https://github.com/zephyrproject-rtos/zephyr/issues/21995
Fix PR: https://github.com/zephyrproject-rtos/zephyr/pull/52012

2. There is also a Control Procedure failure with reason 0x28 instant passed due to a probable limitation in controller when it comes to using/updating the instant to a future event count when under large number of enqueued data PDUs to be transmitted.

Summary:
- #21995 needs to be addressed for multiple connection support with Data Length Updates.
- LLCP implementation needs investigation, to not fail with instant passed .

@kruithofa @thoh-ot @erbr-ot FYI.

cvinayak, 2022-12-15:
@ahmedmoheb-nordic and @jori-nordic  Please try the changes in this PR https://github.com/zephyrproject-rtos/zephyr/pull/53095.

And with `CONFIG_BT_LL_SW_LLCP_LEGACY=y` in your test with the changes in the PR, connection disconnection due to instant passed should also be fixed.

jori-nordic, 2022-12-15:
@cvinayak You're right. Cherry-picking/applying the changes from 53095 and adding the config fixes the test.
Note that the patch alone doesn't make this particular issue disappear, only adding the `LEGACY` cfg fixes it.
Here's the branch: https://github.com/jori-nordic/zephyr/commits/reproduce-53007-fixed

ahmedmoheb-nordic, 2022-12-19:
> @jori-nordic @ahmedmoheb-nordic Can you check if PR #52012 fixes this issue? You'll have to add the following lines to the prj.conf files:
> 
> ```
> CONFIG_BT_CTLR_ADVANCED_FEATURES=y
> CONFIG_BT_CTLR_CONN_EVENT_LEN_UPDATE=y
> ```

@kruithofa Yes, applying changes in PR https://github.com/zephyrproject-rtos/zephyr/pull/52012 fixes the issue

closed 2023-01-20 by jori-nordic