zephyrproject-rtos/zephyr #53019: random: Zephyr enables TEST_RANDOM_GENERATOR when it should not

joerchan, 2022-12-13:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr enables the TEST_RANDOM_GENERATOR in samples and test instead of using the board supported entropy drivers
This configuration should only be used in testing or in samples when no valid entropy generator is available in order to demonstrate proper configuration.

**Describe the solution you'd like**
No sample or test should use the TEST random generator unless the board does not support one.

**Additional context**
There is a build warnings produced in cases where the TEST_RANDOM_GENERATOR is enabled but does not end up being selected due to board or feature entropy default configurations.
The TEST_RANDOM_GENERATOR cannot be enabled when a valid entropy driver is enabled.

Example:
`west build -b native_posix samples/bluetooth/ipsp `

frkv, 2022-12-14:
I would argue that any we potentially need stronger guarantees than disabling usage in wrong contexts. Default-enabling timer as a "pseudo entropy" interface is something that would constitute a security incident if it is was falsely enabled in an end-product.

I want to amend the enhancement-proposal with the following:

If we describe a Kconfig that is naturally set when TEST_RANDOM_GENERATOR enables a fallback "pseudo entropy" is enabled. Its intention (and potential naming) is to signal NOT_MEANT_FOR_PRODUCTION.

To be able to utilize this Kconfig meaningfully, we would then need to add a build-profile you can set (e.g. named PRODUCTION) which will disallow building if the obviously test-specific pseudo entropy source is enabled.

This Kconfig and PRODUCTION build-profile would have multiple use-cases beyond just the related issue, e.g. in case there are options to enable test-keys (not meant for production) or other configurations that disable security critical features

stephanosio, 2022-12-14:
@frkv As noted in https://github.com/zephyrproject-rtos/zephyr/pull/53027#discussion_r1048346691, `a very intrusive warning that is printed every time you run CMake is quite sufficient` should be more than enough in my opinion.

Adding and supporting multiple build profiles is well beyond the scope of this enhancement and should be addressed in a separate issue/feature request.

Also note that we already require `TEST_RANDOM_GENERATOR` to be `n` as part of [`hardenconfig`](https://github.com/zephyrproject-rtos/zephyr/blob/95717520bf4309c7e54fdf6d99975491880fb874/scripts/kconfig/hardened.csv#LL27-L27C22), which should be run for any production firmware

frkv, 2022-12-15:
> Also note that we already require `TEST_RANDOM_GENERATOR` to be `n` as part of [`hardenconfig`](https://github.com/zephyrproject-rtos/zephyr/blob/95717520bf4309c7e54fdf6d99975491880fb874/scripts/kconfig/hardened.csv#LL27-L27C22), which should be run for any production firmware

I would say that `hardenconfig` is doing exactly what I'm asking for, so this is great! I've made a suggestion on default choice of RNG_GENERATOR_CHOICE in the related PR. Please see https://github.com/zephyrproject-rtos/zephyr/pull/53027#discussion_r1049533676

PR #53027 (2022-12-15): random: Clarify the semantics of TEST_RANDOM_GENERATOR
The `TEST_RANDOM_GENERATOR` Kconfig symbol indicates that a non-random
number generator (i.e. a random number generator that is not truly
random) is _allowed_ to be used for testing purposes, and not
necessarily that a non-random number generator is _selected_ -- that is
indicated by the `RNG_GENERATOR_CHOICE` choices that depend on
`TEST_RANDOM_GENERATOR` (i.e. only `TIMER_RANDOM_GENERATOR` as of now).

This commit updates the `TEST_RANDOM_GENERATOR` Kconfig symbol
description to reflect and clarify that.

It also removes the `TEST_RANDOM [... cut]
closed 2022-12-15 by carlescufi