zephyrproject-rtos/zephyr #53033: tests-ci : libraries: encoding: jwt test Failed

hakehuang, 2022-12-14:
**Describe the bug**
    jwt test is Failed on zephyr-v3.2.0-2490-ga1b4896efe46 on mimxrt1170_evk_cm7

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7      --sub-test libraries.encoding
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    
    ```
    START - test_jwt

Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/jwt/src/main.c:52: jwt_tests_test_jwt: (res not equal to 0)
Signing payload
FAIL - test_jwt in 0.133 seconds

        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.2.0-2490-ga1b4896efe46

decsny, 2023-01-23:
This issue is most likely related to the same cause as #52131 and the CAAM since this test enabled the entropy generator

dleach02, 2023-01-31:
Fixed by #54082

closed 2023-01-31 by dleach02