zephyrproject-rtos/zephyr #53035: tests-ci : crypto: rand32: random_ctr_drbg test No Console Output(Timeout)

hakehuang, 2022-12-14:


dleach02, 2022-12-21:
@hakehuang can you bisect this to find where this was broken?

hakehuang, 2022-12-22:
@dleach2 is is below commit

`commit b5708e273b5f34a74e26ec60ac05c759ad0717ce
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Tue Sep 27 09:40:11 2022 -0500

    drivers: entropy: Fix MCUX CAAM Entropy

    Re-enable the CAAM for entropy
    now that the HAL driver has been fixed

    Job descriptors must be accessed coherently
    between CAAM DMA and core.

    The M4 Cores still do not work
    because of mpu/cache/kconfig arch complications,
    disable caam for M4 cores in DTS

    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>

 drivers/entropy/Kconfig.mcux           | 5 ++---
 drivers/entropy/entropy_mcux_caam.c    | 2 +-
 dts/arm/nxp/nxp_rt1160_cm4.dtsi        | 5 +++++
 dts/arm/nxp/nxp_rt1170_cm4.dtsi        | 5 +++++
 tests/drivers/entropy/api/Kconfig.mcux | 9 +++++++++
`

PR #54082 (2023-01-26): drivers: entropy: use non-cache intermediate buffer for RNG
The CAAM hardware needs to read RNG values into a non-cache buffer. Since the contract to Zephyr RNG functions do not require non-cache buffers, we use an intermediate non-cache buffer to retrieve results.

Added a Kconfig to control the size of the intermediate buffer.

Fixes #53035
commit cde15736 (2023-01-26): drivers: entropy: use non-cache intermediate buffer for RNG

The CAAM hardware needs to read RNG values into a non-cache
buffer. Since the contract to Zephyr RNG functions do not
require non-cache buffers, we use an intermediate non-cache
buffer to retrieve results.

Added a Kconfig to control the size of the intermediate buffer.

Fixes #53035
closed 2023-01-26 by mmahadevan108