zephyrproject-rtos/zephyr #53038: tests-ci : net: socket: mgmt test No Console Output(Timeout)

hakehuang, 2022-12-14:
**Describe the bug**
    mgmt test is No Console Output(Timeout) on zephyr-v3.2.0-2490-ga1b4896efe46 on mimxrt1170_evk_cm7

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7      --sub-test net.socket
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    
    ```
    W: You have 1 IPv6 net_if addresses but 2 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0xfffffff8
E: r3/a4:  0x800087ac r12/ip:  0x40460000 r14/lr:  0x30011441
E:  xpsr:  0xa0000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000a58 (main)
E: Halting system

        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.2.0-2490-ga1b4896efe46

hakehuang, 2022-12-14:
@dleach02 , @mmahadevan108 , in rt11xx series almost all net socket related cases have the same fault.

dleach02, 2022-12-21:
can we bisect these...

hakehuang, 2022-12-22:
git bisect found the issue is same root cause of #53035 with below commit

`
commit https://github.com/zephyrproject-rtos/zephyr/commit/b5708e273b5f34a74e26ec60ac05c759ad0717ce
`

closed 2022-12-22 by hakehuang