zephyrproject-rtos/zephyr #53043: Bluetooth: Peripheral misses notifications from Central after bonding

ahmedmoheb-nordic, 2022-12-14:
**Application flow**
1. Central connects to the peripheral
2. At the central side, if this is the first connection, set security level to "Level 2: Encryption and no authentication (no MITM)."
3. At both sides (central and peripheral), if this the first connection, discover the other end primary service and subscribe for notification
4. Both sides start to send notifications in both ways
5. After about 70 seconds, peripheral disconnects and start to advertise again to allow the central to connect to it again

**Describe the bug**
Starting from the second connection round
- At both sides (Central and Peripheral), CCC (Client Characteristic Configuration) callback is called to notify the application that a peer is subscribed for notifications
- Both sides send notifications and no error code is returned while sending notifications (so each end think that other end is still subscribed for notifications)
- Central receives notifications sent by the peripheral as expected
- At the peripheral side:
  - Application receives a callback the it isn't subscribed for notifications anymore
  - Notifications sent by the central aren't received
- CCC callback at the central side isn't called to notify the application that there is no peer currently subscribed

**Workaround**
- Peripheral re-discover again the central's service and re-subscribe for notification

**To Reproduce**
Steps to reproduce the behavior:
1. Pull this commit 1f0fb3f9d541d958451d2f605408ae0185f6a3df
2. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Expected behavior**
Behavior at both sides should be consistent
- Either subscribing again is required
- Or re-subscription isn't required

**Impact**
- Even if the peripheral attempt to re-subscribe again, some notifications can be missed if the central starts to send notification based on the CCC callback event at its side

**Logs and console output**
Please check the following log
[log.log](https://github.com/zephyrproject-rtos/zephyr/files/10225921/log.log)
1. The second connection round starts at line 128
2. At line 146, Central received CCC callback that a device had subscribed for notifications
3. At line 148, Peripheral received a callback that it isn't subscribed anymore
 
[image]

jori-nordic, 2023-01-25:
unreproducible on main.
What happens now is that the central _does_ get the CCC write callback indicating the peer is not subscribed anymore. It then stops sending notifications (as per the test design).

jori-nordic, 2023-01-26:
Figured it out. It's actually hitting #53883 .

The peripheral gets the `->subscribe()` cb because the two devices are bonded and so the quirk to re-subscribe fires but fails with an `BT_ATT_ERR_INSUFFICIENT_ENCRYPTION`.

The patch that fixes #53883 also fixes this issue, so closing as duplicate.

closed 2023-01-26 by jori-nordic