zephyrproject-rtos/zephyr #53046: Bluetooth: Failed to set security level for the second connection

ahmedmoheb-nordic, 2022-12-14:
An HCI error is generated with the following scenario
- Start a scan procedure and connect to the first peripheral
- From the 'connected' callback, set the security level for the current connection and start a new scan procedure
- When the second device is found, start a new connection to the second peripheral

You should be able to navigate to "**~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress**" and run the test script that through "**./tests_scripts/conn_stress.sh**" that should build and run the applications using BabbleSim

Edit: Add a screen shot
[image]

jori-nordic, 2022-12-16:
doesn't happen any more after rebasing on main. Closing.

closed 2022-12-16 by jori-nordic

kruithofa, 2022-12-19:
As per comment from @jori-nordic fixed by PR#51300