zephyrproject-rtos/zephyr #53058: Sensor driver (lps22hh example)

sevirip, 2022-12-14:
**Is your feature request related to a problem? **
Hi,

I wanted to use lps22hh pressure sensor for my project. I needed an hight precision to be able to calculate height.

The issue is that we the driver don't use include low_noise frequency and the filter 

**Describe the solution you'd like**
The solution is to include low noise frequency in the driver and include the possibility to use the filter. I think there is a lot of pressure sensor that have this kind of option. There is also an error in the driver, the driver don't include 200Hz, because in this for loop , +1 is missing.

for (odr = 0; odr < ARRAY_SIZE(lps22hh_map); odr++) {
		if (freq == lps22hh_map[odr]) {
			break;
		}
	}
i think it's ARRAY_SIZE(lps22hh_map)+1 instead of ARRAY_SIZE(lps22hh_map) only 

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

Best regards,

Pierre-Yves

carlescufi, 2022-12-14:
@sevirip thanks for the report! Why not open a Pull Request yourself with the suggested fix?

sevirip, 2022-12-14:
Hi 
Indeed i will open a Pull request !

avisconti, 2024-03-18:
@sevirip do you still have in plan to add a PR for this?

avisconti, 2024-04-04:
I'm closing the issue. Pls feel free to re-open it just in case.

avisconti, 2024-04-04:
just want to close as "not planned"

avisconti, 2024-04-04:
feel

closed 2024-04-04 by avisconti