zephyrproject-rtos/zephyr #53059: Bluetooth: peripheral main loop task starvation

ahmedmoheb-nordic, 2022-12-14:
**Application flow**
1. Central connects to 2 peripherals
2. At the central side, set security level to "Level 2: Encryption and no authentication (no MITM)."
3. At both sides (central and peripheral),  discover the other end primary service and subscribe for notification
4. Both sides start to send notifications in both ways

**Describe the bug**
- At the first connected device side, while trying to send a GATT notification to the central, a call to `bt_gatt_notify()` takes a lot of time (~7 seconds).
- Notifications aren't received the central side
- After 5 seconds from the establishing the connection, a disconnect event is received at the central side
- After another 4 seconds from receiving the disconnect event at the central side, a disconnect event is received at the peripheral side (the moment at which the previous call to `bt_gatt_notify()` returns)

**To Reproduce**
Steps to reproduce the behavior:
1. Pull this commit 46b788e34b28ccc7ff915a95d10f72574f930bee
2. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Logs and console output**
Please check the attached log 
[log.log](https://github.com/zephyrproject-rtos/zephyr/files/10227778/log.log)

1. At line 139, the last print from device no.1 (abbreviated `d_01`)
2. At line 223, a disconnect event is received at the `d_01` end
3. At line 226, a timer value is printed to measure the amount of time consumed by the last call to `bt_gatt_notify()`

jori-nordic, 2023-01-09:
This again sounds like a controller issue, the setup/symptoms seem similar to the last issues you opened that turned out to be controller. I'll take a look later.

jori-nordic, 2023-01-25:
unreproducible on main (after ctlr fixes).

closed 2023-01-25 by jori-nordic