zephyrproject-rtos/zephyr #53062: allow additional extra_post_build_commands to be executed before any other post_build_commands

rlezuo1, 2022-12-14:
**Is your enhancement proposal related to a problem? Please describe.**
We need to include into resulting binaries timestamps and key-material.

- We could generate timestamp during the build, that however triggers compile & link whenever the timestamp changes.
When using timestamps including seconds that is basically every time, so you can never double check that the build is "completed".

- When adding key-material you do not want to store it the repository at all or want different key-material for debug and release builds. Probably the development team does not even know the release key-material. Our non-zephyr build process patches the resulting ELF file (replacing development with release key-material).

**Describe the solution you'd like**

- hooks to modify the final ELF (i.e. zephyr.elf) with custom scripts
- BEFORE any other CONFIG_BUILD_OUTPUT_* (i.e. hex file includes the changes done to the ELF)

**Describe alternatives you've considered**
There already is extra_post_build_commands, but it is run AFTER the CONFIG_BUILD_OUTPUT_* targets, i.e. the binary patching must be implemented for ELF, HEX and BIN in parallel.

**Additional context**
commit ecdea1c30c implements custom hooks, but only AFTER,

marc-hb, 2023-10-31:
Why not just turn on timestamps with a new Kconfig? Something like:

```
builder$ west build -- -DCONFIG_TIMESTAMPS=y
```

arbrauns, 2023-10-31:
I can't find a `CONFIG_TIMESTAMPS` anywhere in the zephyr tree, are you suggesting creating it in a custom Kconfig? If so, what should its effect be and how would it help with the problem at hand?

marc-hb, 2023-10-31:
> are you suggesting creating it in a custom Kconfig?

Yes

> If so, what should its effect be

You sound like you already have a way to break build reproducibility and incremental builds. I'm merely suggesting you don't enable it by default but only when you need it.

closed 2024-11-14 by nordicjm

marc-hb, 2024-11-14:
> EDIT https://docs.zephyrproject.org/latest/develop/west/extensions.html

Elaborating on this and on @nordicjm's recent comment: you could probably create some new, custom `west post_process` extension that:

1. runs `west build` as usual
2. picks up the "final" zephyr.elf and does all the processing you want.

You could also try to re-use `west sign`:
https://docs.zephyrproject.org/latest/develop/west/sign.html

-----------------

For the record, something similar was submitted (and as of now: rejected) in:
- #76248