zephyrproject-rtos/zephyr #53088: Unable to chage initialization priority of logging subsys

dkalowsk, 2022-12-14:
**Describe the bug**

The current design of subsys/logging has a hard coded POST_KERNEL initialization value of 0.  This forces some drivers to be initialized in earlier PRE_KERNEL stages if the system needs them for logging output.

**Expected behavior**

The subsys/logging to have a mechanism to set the initialization priority at compile time via a Kconfig, similar to other Zephyr devices.

**Impact**

Mostly an annoyance.

**Environment (please complete the following information):**

N/A

**Additional context**

dkalowsk, 2022-12-14:
Resolved with https://github.com/zephyrproject-rtos/zephyr/pull/52994.  Issue filed to get backport to 2.7 enabled

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".

PR #53089 (2022-12-20): [Backport v2.7-branch] log_core: Add Kconfig symbol for init priority
Backport 1675d49b4cba95a529b1126d74723a3220ee3c7e from #52994

Fixes #53088
closed 2023-05-17 by kartben