zephyrproject-rtos/zephyr #53101: esp32: The startup code hangs after reboot via sys_reboot(...)

rillbert, 2022-12-15:
**Describe the bug**
After rebooting the system by calling `sys_reboot(...)` the application does not come up. Instead, the startup code hangs in an infinite loop.

My tests indicate that the problem surfaces within the `rtc_clk_cal_internal(...)` method in `rtc_time.c`.

It seems that the code 
```
    /* Wait for calibration to finish up to another us_time_estimate */
    int timeout_us = us_time_estimate;
    while (!GET_PERI_REG_MASK(TIMG_RTCCALICFG_REG(0), TIMG_RTC_CALI_RDY) &&
            timeout_us > 0) {
        timeout_us--;
        esp_rom_delay_us(1);
    }
```
always times out when coming up after a `sys_reboot(...)` call. It does however, work when rebooting by physically pressing the BOOT button on the devkit.

The timeout above, leads to an infinite do/while loop in the `void select_rtc_slow_clk(...)` method within `clk.c` and thus the startup halts.

I have found three ways to work around this at the moment:

1. Set `CONFIG_ESP32_RTC_CLK_CAL_CYCLES=0` in the prj.conf file. This prevents the affected code path to be executed.
2. increase the timeout in the above code snippet to `int timeout_us = us_time_estimate * 10;`. This apparently gives the calibration enough time to finish.
3. replace the code in `sys_arch_reboot(...)` to call a stripped-down version of `esp_restart_noos_dig()` instead of the current call to `esp_restart_noos()`.

I don´t know the root cause to this behavior, my guess is that the RTC is not properly reconfigured after the reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. flash an application that reboots itself by calling `sys_reboot(...)`
2. run the application

**Impact**
Reboot using the zephyr API is currently broken.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain Zephyr SDK 0.15.2
 - Zephyr branch: main
- Target: esp32 (DevkitC v4)
 - Bootloader: mcuboot

sylvioalves, 2023-01-23:
This was fixed already by https://github.com/zephyrproject-rtos/hal_espressif/pull/172

closed 2023-01-23 by sylvioalves

abdulayubzoi, 2023-04-04:
Hi @sylvioalves 

I'd like to re-open this issue, for ESP32C3. The behavior is the same, i.e. the startup code hangs after reboot via sys_reboot() call. In the PR [https://github.com/zephyrproject-rtos/hal_espressif/pull/172](https://github.com/zephyrproject-rtos/hal_espressif/pull/172) it is mentioned that:

> ESP32C3 needs to stick with default esp_rom_delay_us as it uses custom systimer, which is not ready yet during clock calibration.

I'm not sure what this means in regard to fixing the same issue for ESP32C3. Could you elaborate?

I have tested the issue with 2 environments:

**Environment 1:** _(my current "stable" environment)_
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.23.3
- Toolchain Zephyr SDK 0.15.0
- Zephyr branch: v3.1.0
- Target: esp32c3 (DevkitM-1)
- Bootloader: mcuboot

**Environment 2:**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.25.0
- Toolchain Zephyr SDK 0.16.0
- Zephyr branch: v3.3.0
- Target: esp32c3 (DevkitM-1)
- Bootloader: mcuboot

sylvioalves, 2023-04-04:
@abdulayubzoi, would check this fix in your environment? https://github.com/zephyrproject-rtos/hal_espressif/pull/187

abdulayubzoi, 2023-04-14:
> @abdulayubzoi, would check this fix in your environment? [zephyrproject-rtos/hal_espressif#187](https://github.com/zephyrproject-rtos/hal_espressif/pull/187)

Hi @sylvioalves , after extensive testing, I can say that the fix does work in my environment. Thanks a lot!