zephyrproject-rtos/zephyr #53123: Cannot run a unit test on Mac OSX with M1 Chip

ClaytonNorthey92, 2022-12-16:
When I attempt to build the `rb_tree` unit tests on my Mac Laptop, with an M1 Chip, I get the following build error:
```
zephyr/tests/unit/rbtree/main.c:280:1: error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a segment and section separated by a comma
ZTEST_SUITE(rbtree_api, NULL, NULL, NULL, NULL, NULL);
^
zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:220:15: note: expanded from macro 'ZTEST_SUITE'
        static const STRUCT_SECTION_ITERABLE(ztest_suite_node,                                     \
                     ^
zephyr/include/zephyr/toolchain/common.h:215:2: note: expanded from macro 'STRUCT_SECTION_ITERABLE'
        __in_section(_##struct_type, static, name) __used __noasan
        ^
zephyr/include/zephyr/toolchain/gcc.h:178:31: note: expanded from macro '__in_section'
#define __in_section(a, b, c) ___in_section(a, b, c)
                              ^
zephyr/include/zephyr/toolchain/gcc.h:175:25: note: expanded from macro '___in_section'
        __attribute__((section("." Z_STRINGIFY(a) 
```

Additional Information
* I am using commit `a337bfc3b319825788b8e3c4a4764f8b8098d22e`
* Apple M1 (ARM-based) Chip

**To Reproduce**
1. Clone the repository
2. Run `west init`
3. Run `west update`
4. Run `source zephyr-env.sh`
5. Run `./scripts/twister -W -v -k --testsuite-root tests/unit/rbtree/`

**Expected behavior**
I would expect the tests to build and run, especially unit tests.

**Impact**
I wanted to test and experiment with the `rb_tree` data structure locally, prior to using on a microcontroller. I wanted to do so through unit testing, I am not able to do this.

**Logs and console output**
```
zephyr/tests/unit/rbtree/main.c:280:1: error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a segment and section separated by a comma
ZTEST_SUITE(rbtree_api, NULL, NULL, NULL, NULL, NULL);
^
zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test_new.h:220:15: note: expanded from macro 'ZTEST_SUITE'
        static const STRUCT_SECTION_ITERABLE(ztest_suite_node,                                     \
                     ^
zephyr/include/zephyr/toolchain/common.h:215:2: note: expanded from macro 'STRUCT_SECTION_ITERABLE'
        __in_section(_##struct_type, static, name) __used __noasan
        ^
zephyr/include/zephyr/toolchain/gcc.h:178:31: note: expanded from macro '__in_section'
#define __in_section(a, b, c) ___in_section(a, b, c)
                              ^
zephyr/include/zephyr/toolchain/gcc.h:175:25: note: expanded from macro '___in_section'
        __attribute__((section("." Z_STRINGIFY(a) 
```

carlescufi, 2022-12-16:
Thanks for reporting this @ClaytonNorthey92. This is indeed a bug. I am working on getting @tejlmand access to an Apple Silicon Mac to debug this.

ClaytonNorthey92, 2022-12-16:
awesome, thanks for your help @carlescufi and @tejlmand , if there is anything I can do to help, please don't hesitate to reach out.  I can provide further supplemental info if needed (since I am typing this on an M1 😆 )

carlescufi, 2022-12-19:
No worries @ClaytonNorthey92, will keep you updated here. CC @cfriedt.

cfriedt, 2022-12-19:
@ClaytonNorthey92 - unit tests are run under `native_posix`. MacOS is not yet supported under `native_posix` in Zephyr because it would require:

* creating a new LLVM Linker target that
  - parses GNU linker scripts
  - pretends it is building an ELF binary, while silently building a Mach-O binary
* adding python support for handling Mach-O binaries as if they were ELF binaries
* potentially other hacks

For the time being I run native_posix builds on a Linux machine.

carlescufi, 2022-12-19:
> @ClaytonNorthey92 - unit tests are run under `native_posix`. MacOS is not yet supported under `native_posix` in Zephyr because it would require:

No they are not @cfriedt. Unittests are different, they don't use `native_posix`, they are just a regular "native" CMake project that builds a regular app without using the Zephyr build system.

cfriedt, 2022-12-19:
> No they are not @cfriedt. Unittests are different, they don't use `native_posix`, they are just a regular "native" CMake project that builds a regular app without using the Zephyr build system.

Oh.. does it assume that the binary is ELF?

cfriedt, 2022-12-19:
> No they are not @cfriedt. Unittests are different, they don't use `native_posix`, they are just a regular "native" CMake project that builds a regular app without using the Zephyr build system.

Let's start with the following brokem assumptions then:

* the binary format supports ELF-style section names
* the binary format supports lengthy section names
* the linker knows anything about GNU linker scripts (e.g. section ordering)

carlescufi, 2022-12-19:
@cfriedt but the thing is that Unit Tests are native macOS applications that do not use things like custom linker scripts or any other features of the Zephyr build system. They are just regular CMake portable applications that can be built with any compiler on any operating system.

cfriedt, 2022-12-19:
> @cfriedt but the thing is that Unit Tests are native macOS applications that do not use things like custom linker scripts or any other features of the Zephyr build system. They are just regular CMake portable applications that can be built with any compiler on any operating system.

Well, I mean two hackish workarounds would be to either

* ignore the section attribute for unit tests (maybe there is a Kconfig / macro for that)
* use a constant section name / segment name 

Otherwise we will be limited by the 16-character limit imposed by Mach-O.

Note: this is not a new issue by any means and there have already been many rejected PRs in attempt to circumvent.

I agree though - conceptually, we just want a simple C application to run. However, we also implicitly require ELF everywhere.

ClaytonNorthey92, 2022-12-21:
@cfriedt and @carlescufi thank you for your thoughtful discussion.  Since this bug is low priority, do you mind if I work on a fix? 

From what I understand, the section attribute is invalid for the "mach-o" binary format, and when building for my mac, that is the type of executable file that is built.  So, since the section attributes are an invalid format for mach-o, the build fails.

Would a non-hackish solution be to detect if we're building a mach-o binary, then use a different section name?

Also, I don't think this is a bad option either, although it was coined as "hackish"
> ignore the section attribute for unit tests (maybe there is a Kconfig / macro for that)

cfriedt, 2022-12-22:
> @cfriedt and @carlescufi thank you for your thoughtful discussion.  Since this bug is low priority, do you mind if I work on a fix?

 Please do!

ClaytonNorthey92, 2022-12-22:
Thanks @cfriedt , based on my last message, which approach would you like to see me take (if any of those)?  Does this sound feasible?
> Would a non-hackish solution be to detect if we're building a mach-o binary, then use a different section name?

I could also open multiple PRs with different options, for us to evaluate if that is better.

cfriedt, 2022-12-22:
> Thanks @cfriedt , based on my last message, which approach would you like to see me take (if any of those)?  Does this sound feasible?
>
> I could also open multiple PRs with different options, for us [...]

First start off with an #ifdef in the correct header file. If you are able to craft a new macro, you might run into the 16-char limit in Mach-O section / segment names.

This was mentioned in my ZDS talk.

Good luck!

cfriedt, 2023-03-06:
@ClaytonNorthey92 - no rush. This is a very complex problem. Take care

carlescufi, 2023-11-09:
Perhaps something like this could help:
https://github.com/RIOT-OS/RIOT/pull/6534/files

fabiobaltieri, 2023-11-09:
Well I tried to hack something together by changing the `___in_section` definition but I've bumped into a tiny issue: seems like mach-o uses a segment and section structure, and both are limited in name to 16 characters:

https://github.com/aidansteele/osx-abi-macho-file-format-reference#declaration-6

That's a bit problematic to fit stuff like `_ztest_suite_node`, even without the suffixes we normally use in the Linux build (stuff like `._ztest_unit_test.static.z_ztest_unit_test__lib_base64__test_base64_codec_`).

cfriedt, 2023-11-09:
> Perhaps something like this could help:
> https://github.com/RIOT-OS/RIOT/pull/6534/files

Unfortunately, it's not nearly that simple.

Mach-O segment and section names have a 16-char limit. In order to even get zephyr to compile cleanly, LLVM needs to be modified to generate hashes for section names instead of using the actual section name. Either that or every section name in Zephyr needs to be modified to fit inside 16 chars.

Note: instead of speculating, I would strongly recommend going over the slides here

https://bit.ly/3LnXj8S

Pay careful attention to the links, as they do point to existing branches where work has already been done on e.g. Zephyr, LLVM, etc.

Note: a large portion of ELF and Linux-isms could be removed from Zephyr if we switched to using link-time optimization (LTO). With that, binary format would be irrelevant and Zephyr would be portable, ISO standard C.

fabiobaltieri, 2023-11-09:
Thanks @cfriedt, that illustrates the problem pretty well.

nashif, 2024-03-19:
I have no intentions to fix this and I do not see much value in doing this TBH. This is a deep build system, linker issue so for now at least we need to stick to running those tests on Linux.

closed 2024-03-19 by nashif

lmapii, 2025-11-13:
It would be great if it was at least documented that you can't run unit tests on certain platforms. I spend *hours* trying to get unit tests to work ... In general, the Zephyr documentation lacks host restrictions, just the happy paths are documented (same for the `native_sim` target).

TBH this restriction (and the very limited mocking capabilities with FFF) makes me doubt my decision to pursue testing using ZTEST. It will remain great for integration tests (e.g., using qemu targets if you're not on POSIX), but for unit testing it doesn't seem mature enough.

lmapii, 2025-11-14:
It's this one https://github.com/zephyrproject-rtos/zephyr/pull/99368