zephyrproject-rtos/zephyr #53124: cmake: incorrect argument passing and dereference in zephyr_check_compiler_flag() and zephyr_check_compiler_flag_hardcoded()

smrtos, 2022-12-16:
Below calling sequence seems incorrect.

https://github.com/zephyrproject-rtos/zephyr/blob/caf309c6f6ca5df4f9789c6a33ec3738d202dd5e/cmake/modules/extensions.cmake#L1902

https://github.com/zephyrproject-rtos/zephyr/blob/caf309c6f6ca5df4f9789c6a33ec3738d202dd5e/cmake/modules/extensions.cmake#L1014-L1017

https://github.com/zephyrproject-rtos/zephyr/blob/caf309c6f6ca5df4f9789c6a33ec3738d202dd5e/cmake/modules/extensions.cmake#L1119-L1125

The purpose of the check sequence is to have variable like `check_xxx` set to `0` or `1` in the first calling context.

The issue with above calling sequence can be illustrated with below MRE (Minimal Reproduce Example):

```
function(zephyr_check_compiler_flag_hardcoded lang option check exists)
  set(check 0 PARENT_SCOPE)
endfunction()

function(zephyr_check_compiler_flag lang option check)
  message(STATUS 11111: ${check})
  zephyr_check_compiler_flag_hardcoded(C "" check exists)
  message(STATUS 22222: ${check})
  set(check ${check} PARENT_SCOPE)
  return()
endfunction()

function(target_link_libraries_ifdef feature_toggle target item)
  message(STATUS ${feature_toggle}= ${${feature_toggle}})
endfunction()

set(check "check_xxx")
zephyr_check_compiler_flag(C "" ${check})
target_link_libraries_ifdef(${check} dummy_target dummy_scope dummy_option)

```
The output is:

```
-- 11111:check_xxx
-- 22222:0
-- 0=

```

It should be fixed like this:

```
function(zephyr_check_compiler_flag_hardcoded lang option check exists)
  set(${check} 0 PARENT_SCOPE)   # <=============== here, ${check} is used instead
endfunction()

function(zephyr_check_compiler_flag lang option check)
  message(STATUS 11111: ${check})
  zephyr_check_compiler_flag_hardcoded(C "" check2 exists) # <============= here, check2 is used instead
  message(STATUS 22222: ${check})
  set(${check} ${check2} PARENT_SCOPE) # <============= here, check2 is used instead
  return()
endfunction()

function(target_link_libraries_ifdef feature_toggle target item)
  message(STATUS ${feature_toggle}= ${${feature_toggle}})
endfunction()

set(check "check_xxx")
zephyr_check_compiler_flag(C "" ${check})
target_link_libraries_ifdef(${check} dummy_target dummy_scope dummy_option)

```

The output is:
```
-- 11111:check_xxx
-- 22222:check_xxx
-- check_xxx=0

```

stephanosio, 2023-02-08:
@tejlmand Are you able to look into this issue for v3.3? Does the proposed fix above sound reasonable?

nashif, 2023-02-21:
@tejlmand  is this really a high priority bug?

tejlmand, 2023-03-01:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/55301

yperess, 2023-03-03:
https://github.com/zephyrproject-rtos/zephyr/pull/55301 broke my C++ project

I'm now seeing:
```
cc1plus: error: command-line option '-Wno-pointer-sign' is valid for C/ObjC but not for C++ [-Werror]
cc1plus: error: '-Werror=' argument '-Werror=implicit-int' is not valid for C++ [-Werror]
```
I added some logging to verify and it seems that several call sites of `zephyr_check_compiler_flag` were not updated to account for the extra layer of dereferencing of `check`.

yperess, 2023-03-03:
I'm setting the priority to high because this was backported already and is breaking builds.

yperess, 2023-03-03:
https://github.com/zephyrproject-rtos/zephyr/pull/55301/files#diff-099f8d126fd2a4d31179861b2819d9f607cadd14602968912448d7e5d8248b6cR1021 is currently stomping me. If `zephyr_check_compiler_flag_hardcoded` sets `exists` to `1` then `${check}` is set to `0`. This means that `set(${check} ${${check}} PARENT_SCOPE)` resolves to `set(0)`. I can't figure out if there's a use case I missed.

yperess, 2023-03-03:
https://github.com/zephyrproject-rtos/zephyr/pull/55418 fixes the issue for me, but I wasn't able to verify all the other call sites

tejlmand, 2023-03-06:
> I added some logging to verify and it seems that several call sites of zephyr_check_compiler_flag were not updated to account for the extra layer of dereferencing of check.

and they were not suppose to.
The old error was that an internal variable name called `check` was used, and then in parent scope the variable of same name `check` was updated.
If caller, by luck, was also using the var name `check`, then everything worked as expected, but if caller used another name, like `other_name`, like here:
https://github.com/zephyrproject-rtos/zephyr/blob/f5bf1cdddabbe1e2e3cdc7ef150504db6ad266e1/cmake/modules/kernel.cmake#L144

in that case, a `check` variable would be created containing the result, and not `toolchain_is_ok`.

but I have afterwards noticed a couple of wrong use of `check` vs `${check}` which by pure luck did not hide the original argument, for example like here:
https://github.com/zephyrproject-rtos/zephyr/blob/f5bf1cdddabbe1e2e3cdc7ef150504db6ad266e1/cmake/modules/extensions.cmake#L1963-L1970
```
# Creates an identifier, let's call it `foo`, meaning value of `check=foo`
string(MAKE_C_IDENTIFIER ${check_identifier} check) 
  
...
# Check the flag, store it in a local variable named as the value of `check`, i.e. in a variable named `foo`.
# Note, old code would accidentally always return the result in a variable named `check` and not in the variable which is pointed to by `check`, in this case `foo`.
zephyr_check_compiler_flag(C "" ${check}) 
...
# This code check the value of `check`, and not the value of `foo` (`${check}`), which it is supposed to.
# It only worked because it was named `check` which by luck was the same name used in called function.
target_link_libraries_ifdef(${check} ${target} ${scope} ${option}) 
```

> https://github.com/zephyrproject-rtos/zephyr/pull/55301 broke my C++ project

can you reproduce this in the Zephyr tree, or provide a small snippet / example reproducing this ?
It would be nice for reproducing and verifying my above findings.

@yperess I know that it can sometimes be hard to judge whether a bug is fixed or not vs. fixed but exposes other (hidden) bugs in code.
In this case, the original issue was indeed fixed by #55301, but there seems to be cases who was calling `zephyr_check_compiler_flag()` wrongly, so would be nice in such case to have a dedicated issue filed.

commit 04a27651 (2023-03-01): cmake: fix variable de-referencing in zephyr_check_compiler_x functions

Fixes: #53124

Fix de-referencing of check and exists function arguments by correctly
de-referencing the argument references using `${<var>}`.
commit 2b6e40e8 (2023-03-03): cmake: fix variable de-referencing in zephyr_check_compiler_x functions

Fixes: #53124

Fix de-referencing of check and exists function arguments by correctly
de-referencing the argument references using `${<var>}`.

(cherry picked from commit 04a27651ea7402ecd24fa907983416ed3fc26a12)
commit b02d34b8 (2023-03-13): cmake: fix variable de-referencing in zephyr_check_compiler_x functions

Fixes: #53124

Fix de-referencing of check and exists function arguments by correctly
de-referencing the argument references using `${<var>}`.

(cherry picked from commit 04a27651ea7402ecd24fa907983416ed3fc26a12)
closed 2023-03-06 by stephanosio