zephyrproject-rtos/zephyr #53126: Define settings in sysbuild Kconfig and and propagate those to image(s) Kconfig

tejlmand, 2022-12-16:
Part of: #41640 

When using sysbuild to control multiple images, then it is often necessary to share a setting across builds.
And if a setting is shared, then ensure if a user reconfigures the setting, then this information is updated across images.

It is therefore required that it will be possible to define settings in sysbuild Kconfig and and then propagate those settings to image(s) Kconfig.
This should result in those settings becoming inaccessible by users, i.e. they should be promptless.

When implementing this, then the following must be considered:
- A setting must be shared to all images being build for same board / SoC (and same core on the SoC).
- A setting must be shared to all images being build for same board / SoC but targetting different cores on the SoC.
- A setting must go to all images regardless of the board used.

nordicjm, 2023-01-25:
@de-nordic this seems like a good candidate for sharing information in builds between mcuboot and target images which can then be used to check if e.g. an update is too large to fit for a given mcuboot image.

de-nordic, 2023-01-26:
> @de-nordic this seems like a good candidate for sharing information in builds between mcuboot and target images which can then be used to check if e.g. an update is too large to fit for a given mcuboot image.

Yes, this could allow us to better integrate builds of MCUmgr enabled apps with MCUboot and impose the MCUboot configuration on the former.
Maybe this can also help us with building apps for Direct XIP?

PR #55871 (2023-05-25): sysbuild: generate .config.sysbuild for controlling build settings
Sysbuild now generates a .config.sysbuild config file which specifies
settings controlled by sysbuild.
Any setting specified in this .config will overrule user provided
setting, and a warning will be raised if the sysbuild controlled value
is different from the value specified by the user.

This has the following benefits:
- Allow sysbuild to control any build setting without adjustments to
the existing Kconfig tree
- Allow sysbuild to adjust settings based on knowledge regarding enabled
images / bootloaders.
- Cleanup CMake co [... cut]