zephyrproject-rtos/zephyr #53129: Build fails on ESP32 when enabling websocket client API

RicArch97, 2022-12-16:
**Describe the bug**
When enabling `CONFIG_WEBSOCKET_CLIENT`, compilation fails on ESP32 presumably due to `CONFIG_BASE64` being enabled by `CONFIG_WEBSOCKET_CLIENT`, of which the functions are already defined in the Espressif HAL module.

Target platform: Xtensa esp32

**To Reproduce**
Enable `CONFIG_WEBSOCKET_CLIENT` in `prj.conf` of any project, and compile for `esp32` with `west build -b esp32 <project>`

**Expected behavior**
Build for esp32 to succeed with the websocket client API. 

**Impact**
Currently stalls development. Could write my own implementation of websocket client if necessary

**Logs and console output**
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-g++  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/wifi/libsubsys__net__l2__wifi.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/net/lib/http/libsubsys__net__lib__http.a  zephyr/subsys/net/lib/websocket/libwebsocket_client.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/wifi/libdrivers__wifi.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0"  -L/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lm  -Wl,-lc  -L"/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf"/lib/.  -Wl,-lgcc  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.redefined.ld  -T/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -L/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/zephyr/esp32/../blobs/lib/esp32 && cd /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr && /nix/store/vh76h64xifdiylk1dfc9s6mgpzazngxs-cmake-3.24.3/bin/cmake -E echo
/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(base64.c.obj): in function `base64_encode':
/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/components/wpa_supplicant/src/utils/base64.c:166: multiple definition of `base64_encode'; zephyr/libzephyr.a(base64.c.obj):/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/lib/os/base64.c:62: first defined here
/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(base64.c.obj): in function `base64_decode':
/home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/components/wpa_supplicant/src/utils/base64.c:188: multiple definition of `base64_decode'; zephyr/libzephyr.a(base64.c.obj):/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/lib/os/base64.c:126: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /nix/store/vh76h64xifdiylk1dfc9s6mgpzazngxs-cmake-3.24.3/bin/cmake --build /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build
```

**Environment (please complete the following information):**
 - OS: Linux (NixOS)
 - Toolchain: Zephyr SDK 0.15.2, xtensa-espressif_esp32_zephyr-elf
 - Commit SHA: `b8f8c3c10eda6d1529d0e616b24ba9719efab2c8`

sylvioalves, 2022-12-16:
I have tried building `samples/net/wifi` adding `CONFIG_WEBSOCKET_CLIENT=y` without errors. can you share more info on how to reproduce this?

RicArch97, 2022-12-16:
Using `samples/net/wifi` as example; Add `CONFIG_WEBSOCKET_CLIENT=y` to `samples/net/wifi/prj.conf`, then build with `west build -b esp32 samples/net/wifi`

My full build output of it:
```
$ west build -p auto -b esp32 samples/net/wifi          
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/samples/net/wifi
-- Found Python3: /nix/store/s16n3im3ajnbzamp6yzzp15cwxqn8pbr-python3-3.10.8/bin/python3.10 (found suitable exact version "3.10.8") found components: Interpreter 
-- Cache files will be written to: /home/ricardo/.cache//zephyr
-- Zephyr version: 3.2.99 (/home/ricardo/Documents/Programming/zephyr-workspace/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: esp32
-- Found host-tools: zephyr 0.15.2 (/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2)
-- Found Dtc: /nix/store/pq5gybjkrnwpkhcc9zrvf1ib2fyk50xh-dtc-1.6.1/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/boards/xtensa/esp32/esp32.dts
-- Found devicetree overlay: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/samples/net/wifi/boards/esp32.overlay
-- Generated zephyr.dts: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/dts.cmake
/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/zephyr.dts:37.39-44.5: Warning (interrupt_provider): /soc/interrupt-controller@3ff00104: Missing #address-cells in interrupt provider
Parsing /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/Kconfig
Loaded configuration '/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/samples/net/wifi/prj.conf'
Merged configuration '/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/samples/net/wifi/boards/esp32.conf'
Configuration saved to '/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
CMake Warning at /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/CMakeLists.txt:824 (message):
  No SOURCES given to Zephyr library: drivers__ethernet

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build
-- west build: building application
[1/407] Preparing syscall dependency handling

[14/407] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/ricardo/Documents/Programming/zephyr-workspace/zephyr), build: zephyr-v3.2.0-2647-gb8f8c3c10eda
[30/407] Performing configure step for 'EspIdfBootloader'
-- Found Git: /nix/store/1im1jlvkqfzwdwyps9nwychkl638ypz7-git-2.38.1/bin/git (found version "2.38.1") 
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/zgac4ca0gd1zidn558p6gnbznrrn8bdq-zephyr-sdk-0.15.2/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build/esp-idf/build/sdkconfig
-- Adding linker script /home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/ricardo/Documents/Programming/zephyr-workspace/modules/hal/espressif/components/esp_rom/esp32/ld/esp32.rom.api.ld
[... cut ...] phyr-workspace/zephyr/lib/os/base64.c:126: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /nix/store/vh76h64xifdiylk1dfc9s6mgpzazngxs-cmake-3.24.3/bin/cmake --build /home/ricardo/Documents/Programming/zephyr-workspace/zephyr/build
```

Build of `samles/net/wifi` succeeds without having added `CONFIG_WEBSOCKET_CLIENT=y` to `samples/net/wifi/prj.conf`. 

These are all the steps i took to produce this

PR #53222 (2022-12-22): hal_espressif: remove base64 redefinition
WPA supplicant includes custom base64.c file with no need.

Fixes #53129
commit 8a52f9a8 (2022-12-22): hal_espressif: remove base64 redefinition

WPA supplicant includes custom base64.c file with no need.

Fixes #53129
closed 2022-12-22 by carlescufi