zephyrproject-rtos/zephyr #53135: `ROUND_UP` and `ROUND_DOWN` common utility macros not guarded and functionally limited

besmarsh, 2022-12-16:
**Is your enhancement proposal related to a problem? Please describe.**

Miscellaneous utility macros are defined in `zephyr/sys/util.h` and are "usable by the kernel and application code". These include common macros `ROUND_UP` and `ROUND_DOWN`.
As the `zephyr/sys/utils.h` macros are used by the kernel, they are always defined for application code including `zephyr/kernel.h`, but the macro definitions are not guarded, so cause macro redefinition warnings if they are also defined by application code.
Furthermore, the `ROUND_UP` and `ROUND_DOWN` macros defined in `zephyr/sys/util.h` only work with alignments that are a power of 2, which is somewhat limited, and application code wishing to use `ROUND_UP` and `ROUND_DOWN` may well want to do so with alignments that are not a power of 2.

**Describe the solution you'd like**

I'd like to see at least the common macros `ROUND_UP` and `ROUND_DOWN` guarded by `#ifndef`, and maybe other utility macros should be guarded too (note that `MIN` and `MAX` are already guarded).

I'd also like to see either
* these macros made more general (i.e. work for alignments that are not a power of 2) so they are more useful to application code and function as expected, or
* the names of the macros changed to something more specific so as to not interfere with more general application-defined macros with the general names `ROUND_UP` and `ROUND_DOWN`.

**Describe alternatives you've considered**

Namespacing the macros with a Zephyr-specific prefix could also be considered to prevent interference with application-defined macros with the same names.

**Additional context**

I'm interested to hear what would be the preferred course of action for this. Maybe there is another solution I haven't considered that would be fine too!

carlescufi, 2022-12-19:
Hi @besmarsh, thanks for your input. There is an ongoing discussion here: https://github.com/zephyrproject-rtos/zephyr/pull/51963

besmarsh, 2022-12-30:
Thanks @carlescufi. I've read up on the discussion in the various issues and the Discord channel. I understand the rationale for Zephyr to be the central project that defines these common macros and therefore the approach proposed in #51963.

All the discussions I've read on this issue consider the collisions between Zephyr and libraries/modules/HALs used by Zephyr, but not collisions between Zephyr and user application code. I realise that user code should either use the Zephyr-defined macros, guard its own macros with `#ifndef`, or namespace its own if redefining. In most cases (for these "common macros") this is fine.

However I think this issue highlights a separate enhancement, as the `ROUND_UP` and `ROUND_DOWN` macros provided by Zephyr are not as useful as they could be for user applications. This issue could just be for changing the Zephyr-defined `ROUND_UP` and `ROUND_DOWN` macros to work for non-power-of-2 alignments. Is there any reason why this can not/should not be done?

besmarsh, 2023-01-18:
> This issue could just be for changing the Zephyr-defined `ROUND_UP` and `ROUND_DOWN` macros to work for non-power-of-2 alignments. Is there any reason why this can not/should not be done?

Hi @carlescufi, any thoughts? I can open a PR for this if there's no reason not to do it.

besmarsh, 2023-05-21:
> Hi @besmarsh, this was actually discussed and we came up with a conclusion here: https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-3-macro-name-collisions

Thanks @carlescufi, I saw that, but it doesn't address this point:

> I think this issue highlights a separate enhancement, as the ROUND_UP and ROUND_DOWN macros provided by Zephyr are not as useful as they could be for user applications. This issue could just be for [...]

besmarsh, 2023-05-29:
@carlescufi PR opened - https://github.com/zephyrproject-rtos/zephyr/pull/58400

besmarsh, 2024-06-20:
Issue was resolved by #58400

closed 2024-06-20 by besmarsh