zephyrproject-rtos/zephyr #53137: Bluetooth: Controller: HCI 0x45 error after 3rd AD fragment with data > 248 bytes

Tronil, 2022-12-16:
Attempting to set advertising data with the HCI_LE_Set_Extended_Advertising_Data command using the maximum allowed fragmentation size (251 bytes) fails after the 3rd fragment with error 0x45 (BT_HCI_ERR_PACKET_TOO_LONG). If I instead reduce the fragment size down to 248 bytes it works and I can set the entirety of the advertising data.

It looks like we are hitting this case in `ll_adv_aux_ad_data_set()`:
```
		/* Check AdvData overflow */
		if ((sec_len + ad_len_overflow + len) >
		    PDU_AC_PAYLOAD_SIZE_MAX) {
			/* NOTE: latest PDU was not consumed by LLL and
			 * as ull_adv_sync_pdu_alloc() has reverted back
			 * the double buffer with the first PDU, and
			 * returned the latest PDU as the new PDU, we
			 * need to enqueue back the new PDU which is
			 * infact the latest PDU.
			 */
			if (pdu_prev == pdu) {
				lll_adv_aux_data_enqueue(adv->lll.aux, sec_idx);
			}

			return BT_HCI_ERR_PACKET_TOO_LONG;
		}
```
Here, `len` will be 251 so with the addition of `sec_len` and `ad_len_overflow` it can easily be more than 255 (`PDU_AC_PAYLOAD_SIZE_MAX`).

Easy way to reproduce is to run the LL/DDI/ADV/BV-47-C or LL/DDI/ADV/BV-49-C tests in EDTT (added in PR: https://github.com/EDTTool/EDTT/pull/67).

cvinayak, 2022-12-17:
Advertising Data chaining is experimental, does not implement re-combining or AD data overflow in PDUs. A discussion is needed on choice of fragmenting and recombining, as current implementation is not optimal in terms of number of PDUs being used.

https://github.com/zephyrproject-rtos/zephyr/blob/10165b7d79faaec68b6e7a05098505d08fe9f0f1/subsys/bluetooth/controller/Kconfig.ll_sw_split#L202-L216

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".

Thalley, 2026-09-10:
LL/DDI/ADV/BV-45-C and LL/DDI/ADV/BV-52-C still fails, and they reference this issue.