zephyrproject-rtos/zephyr #53147: usb: cdc_acm: log related warning promoted to error

cfriedt, 2022-12-16:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. `twister -i -p degu_evk -s zephyr/tests/posix/common/portability.posix.common.tls.newlib`
2. See error

**Expected behavior**
The test should build and pass.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```shell
/__w/zephyr/zephyr/subsys/usb/device/class/cdc_acm.c
/__w/zephyr/zephyr/subsys/usb/device/class/cdc_acm.c:63:2: error: #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE" [-Werror=cpp]
   63 | #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE"
      |  ^~~~~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): any
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.15.2
 - Commit SHA or Version used: 1173db6f605aca79f38d6fbcf2e634f5dbd02f3f

**Additional context**
It's likely that this has to do with a recent change in logging cc @nordic-krch 

This is currently failing in #52977 even after a rebase.

jfischer-no, 2022-12-16:
This is a kind of regression caused by #53032, degu_evk configuration has to be fixed.

cfriedt, 2022-12-18:
This was *not* fixed by #53149. #52977 is still failing due to the following.

```

/Users/cfriedt/workspace/zephyrproject/zephyr/subsys/usb/device/class/cdc_acm.c:63:2: error: #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE" [-Werror=cpp]
   63 | #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE"
      |  ^~~~~~~
```

> This is a kind of regression caused by #53032, degu_evk configuration has to be fixed.

It seems that is not enough to fix the problem in other places.

cfriedt, 2022-12-18:
This was not* fixed by #53149. #52977 is still failing due to the following.

```
twister -i -p bl654_usb                 -s zephyr/tests/posix/common/portability.posix.common.tls
...
/Users/cfriedt/workspace/zephyrproject/zephyr/subsys/usb/device/class/cdc_acm.c:63:2: error: #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE" [-Werror=cpp]
   63 | #warning "USB_CDC_ACM_LOG_LEVEL forced to LOG_LEVEL_NONE"
      |  ^~~~~~~
```

cfriedt, 2022-12-18:
@stephanosio - thought I should pull you into this one. It just warmed up a bit.

desowin, 2022-12-19:
> is there a fix for this issue that would prevent needing more future fixes in other areas?

A possible "fix" is to just remove the `#warning`. The downside is that the configured CDC ACM log level would be silently ignored.

desowin, 2022-12-19:
It might be that `bl654_usb` is the only one left, or that there are still others. I am running twister locally on all platforms, build only, on `tests/posix/common` from #52977. This will take quite some time, but at least it will give some better indication about failing targets.

closed 2022-12-19 by fabiobaltieri

desowin, 2022-12-19:
For what it is worth, running twister for all boards, i.e.
```
./scripts/twister --jobs 2 -p native_posix -p native_posix_64 -p nrf52_bsim -p qemu_nios2 -p altera_max10 -p esp32s2_franzininho -p intel_adsp_cavs20 -p intel_adsp_cavs20_jsl -p heltec_wifi_lora32_v2 -p esp32s2_saola -p esp_wrover_kit -p esp32 -p intel_adsp_cavs15 -p qemu_xtensa -p esp32_net -p olimex_esp32_evb -p nxp_adsp_imx8x -p nxp_adsp_imx8m -p esp32_ethernet_kit -p nxp_adsp_imx8 -p odroid_go -p intel_adsp_ace15_mtpm -p xt-sim -p intel_adsp_cavs25_tgph -p intel_adsp_cavs25 -p intel_adsp_cavs18 -p qemu_malta -p qemu_malta_be -p acrn_ehl_crb -p acrn -p rpl_crb -p ehl_crb_sbl -p ehl_crb -p qemu_x86_xip -p qemu_x86_virt -p qemu_x86_nokpti -p qemu_x86_tiny -p qemu_x86_tiny@768 -p qemu_x86_lakemont -p qemu_x86_nopae -p qemu_x86 -p qemu_x86_64_nokpti -p qemu_x86_nommu -p qemu_x86_64 -p up_squared -p beaglev_starlight_jh7100 -p gd32vf103v_eval -p mpfs_icicle -p it8xxx2_evb -p hifive1 -p qemu_riscv64_smp -p qemu_riscv64 -p litex_vexriscv -p hifive_unleashed -p hifive_unmatched -p m2gl025_miv -p esp32c3_devkitm -p xiao_esp32c3 -p adp_xc7k_ae350 -p qemu_riscv32_smp -p qemu_riscv32_xip -p qemu_riscv32 -p tlsr9518adk80d -p qemu_riscv32e -p neorv32 -p neorv32@1.6.1 -p gd32vf103c_starter -p longan_nano -p longan_nano_lite -p hifive1_revb -p icev_wireless -p rv32m1_vega_zero_riscy -p rv32m1_vega_ri5cy -p emsdp_em5d -p emsdp_em6 -p emsdp_em9d -p emsdp_em7d_esp -p emsdp -p emsdp_em4 -p emsdp_em7d -p em_starterkit_em7d -p em_starterkit_em7d_v22 -p em_starterkit_em11d -p em_starterkit -p qemu_arc_hs5x -p qemu_arc_hs6x -p qemu_arc_hs_xip -p qemu_arc_hs -p qemu_arc_em -p iotdk -p hsdk -p hsdk_2cores -p nsim_hs_mpuv6 -p nsim_hs5x_smp -p nsim_em -p nsim_em7d_v22 -p nsim_hs -p nsim_hs5x -p nsim_hs_flash_xip -p nsim_sem_mpu_stack_guard -p nsim_hs6x -p nsim_hs_smp -p nsim_hs6x_smp -p nsim_sem -p nsim_hs_sram -p mimx8mm_evk_a53 -p mimx8mm_evk_a53_smp -p qemu_kvm_arm64 -p fvp_base_revc_2xaemv8a -p fvp_base_revc_2xaemv8a_smp_ns -p khadas_edgev -p mimx8mp_evk_a53_smp -p mimx8mp_evk_a53 -p nxp_ls1046ardb -p nxp_ls1046ardb_smp_2cores -p nxp_ls1046ardb_smp_4cores -p fvp_baser_aemv8r -p fvp_baser_aemv8r_smp -p xenvm_gicv3 -p xenvm -p bcm958402m2_a72 -p intel_socfpga_agilex_socdk -p qemu_cortex_a53 -p qemu_cortex_a53_xip -p qemu_cortex_a53_smp -p mimx8mn_evk_a53_smp -p mimx8mn_evk_a53 -p gr716a_mini -p generic_leon3 -p qemu_leon3 -p atsaml21_xpro -p nucleo_f303k8 -p mimx8mm_evk -p pan1782_evb -p olimex_stm32_e407 -p steval_fcu001v1 -p ip_k66f -p stm32l562e_dk -p stm32l562e_dk_ns -p 96b_avenger96 -p stm32f0_disco -p serpente -p mimxrt685_evk_cm33 -p bl653_dvk -p nucleo_f334r8 -p xmc45_relax_kit -p nucleo_l432kc -p stm32g0316_disco -p efm32hg_slstk3400a -p adafruit_feather_nrf52840 -p pico_pi_m4 -p stm32_min_dev_black -p stm32_min_dev_blue -p qemu_cortex_m0 -p nucleo_f411re -p nucleo_f303re -p lpcxpresso55s28 -p mimxrt1170_evk_cm4 -p mimxrt1170_evk_cm7 -p stm32mp157c_dk2 -p nucleo_wb55rg -p sam4e_xpro -p efm32pg_stk3402a -p efm32pg_stk3402a_jg -p nrf52832_mdk -p arduino_portenta_h7_m4 -p arduino_portenta_h7_m7 -p nucleo_f091rc -p stm32h747i_disco_m7 -p stm32h747i_disco_m4 -p frdm_kw41z -p nrf51_ble400 -p mec15xxevb_assy6853 -p adafruit_feather_m0_basic_proto -p disco_l475_iot1 -p stm32f4_disco -p twr_ke18f -p s32z270dc2_rtu0_r52 -p s32z270dc2_rtu1_r52 -p nrf9160dk_nrf52840 -p gd32e103v_eval -p sam4s_xplained -p we_ophelia1ev_nrf52805 -p 96b_argonkey -p legend -p nrf21540dk_nrf52840 -p nucleo_f207zg -p nucleo_h7a3zi_q -p adafruit_itsybitsy_m4_express -p nucleo_l433rc_p -p nucleo_l031k6 -p sensortile_box -p nucleo_f446re -p nrf51dongle_nrf51422 -p mimxrt1015_evk -p nrf52_adafruit_feather -p rak4631_nrf52840 -p nucleo_l452re -p nucleo_l452re_p -p nrf52840_mdk -p nucleo_f302r8 -p nucleo_f030r8 -p nucleo_f030r8@1 -p nucleo_f030r8@2 -p cc1352r_sensortag -p nucleo_l073rz -p arduino_nicla_sense_me -p mps2_an385 -p atsamr21_xpro -p waveshare_open103z -p stm32f3_disco -p stm32f3_disco@B -p stm32f3_disco@E -p olimex_lora_stm32wl_devkit -p ast1030_evb -p nrf52_sparkfun -p qemu_cortex_m3 -p cyclonev_socdk -p fvp_baser_aemv8r_aarch32 -p scobc_module1 -p we_proteus3ev_nrf52840 -p actinius_icarus_som -p actinius_icarus_som_ns -p stm32l1_disco -p nucleo_l152re -p nrf51_vbluno51 -p ubx_evkannab1_nrf52832 -p rddrone_fmuk66 -p zybo -p nucleo_h745zi_q_m4 -p nucleo_h745zi_q_m7 -p stm32f769i_disco -p hexiwear_k64 -p contextualelectronics_abc -p stm32373c_eval -p b_l4s5i_iot01a -p reel_board -p reel_board_v2 -p bl5340_dvk_cpunet -p bl5340_dvk_cpuapp -p bl5340_dvk_cpuapp_ns -p ubx_evkninab1_nrf52832 -p mimxrt1160_evk_cm7 -p mimxrt1160_evk_cm4 -p stm32f072b_disco -p bl652_dvk -p mimxrt1010_evk -p b_u585i_iot02a -p b_u585i_iot02a_ns -p olimex_stm32_h407 -p v2m_beetle -p frdm_kl25z -p arduino_due -p mikroe_clicker_2 -p nucleo_l4a6zg -p mps3_an547 -p mps3_an547_ns -p efr32_radio_brd4104a -p efr32_radio_brd4255a -p efr32_radio_brd4250b -p efr32_radio_brd4180a -p adafruit_itsybitsy_nrf52840 -p nrf5340dk_nrf5340_cpuapp -p nrf5340dk_nrf5340_cpunet -p nrf5340dk_nrf5340_cpuapp_ns -p nucleo_g431rb -p arduino_mkrzero -p bcm958402m2_m7 -p sam_e70b_xplained -p sam_e70_xplained -p cy8ckit_062_ble_m0 -p cy8ckit_062_ble_m0@1.0.0 -p cy8ckit_062_ble_m0@0.0.0 -p cy8ckit_062_ble_m4 -p cy8ckit_062_ble_m4@0.0.0 -p stm32vl_disco -p twr_kv58f220m -p nucleo_f412zg -p efm32gg_slwstk6121a -p 96b_carbon -p adafruit_feather_m0_lora -p mimxrt1024_evk -p atsamd21_xpro -p xiao_ble -p lpcxpresso55s36 -p stm32h735g_disco -p v2 [... cut ...]
INFO    - Saving reports...
INFO    - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

The `twister-out` directory size was 64GiB. I hope there are no more CDC ACM logging related issues, but if there are, then please bear in mind that it is not easy to find them.

fabiobaltieri, 2022-12-19:
> For what it is worth, running twister for all boards, i.e.
>
> ```
> ./scripts/twister --jobs 2 -p native_posix -p native_posix_64 -p nrf52_bsim -p qemu_nios2 -p altera_max10 -p esp32s2_franzininho -p [...]

Would this show up in the biweekly run? (https://github.com/zephyrproject-rtos/zephyr/actions/workflows/twister.yaml?query=event%3Aschedule) Maybe we could try and get better visibility to those.

cfriedt, 2022-12-19:
@desowin - it's a bit scary to me to think that the issue might randomly resurface over again later, whenever CDC_ACM is used as the serial interface. Is there no cleaner solution than to tack on a `Kconfig.board` every time?

desowin, 2022-12-21:
> Would this show up in the biweekly run? (https://github.com/zephyrproject-rtos/zephyr/actions/workflows/twister.yaml?query=event%3Aschedule) Maybe we could try and get better visibility to those.

@PerMac @gopiotr Do you know what is the test coverage? Does it build all samples for all targets or just a selected set? I guess there is not 100% coverage and there is quite some dependency on vendors reporting issues.

desowin, 2022-12-21:
> @desowin - it's a bit scary to me to think that the issue might randomly resurface over again later, whenever CDC_ACM is used as the serial interface. Is there no cleaner solution than to tack on a `Kconfig.board` every time?

This is what you get when the essentially flawed Kconfig configuration is not prevented from the beginning. That is, if you prevent the incorrect configuration years after it was possible to build with such config, you'll keep discovering issues in places you don't expect for some time. However the sooner you start warning/failing builds with incorrect config, the less problems you'll have in total.

PR #53177 (2022-12-19): boards: bl654_usb: disable CDC ACM logging
Disable logging module in CDC ACM to prevent recursive logging loop when CDC ACM is used as serial backend.

Fixes: #53147