zephyrproject-rtos/zephyr #53158: SPIM transaction timeout leads to crash

mnmoron, 2022-12-17:
I have discovered a defect in the `spi_nrfx_spim.c` driver causing hard faults on my nRF5340 based system. This is related to my question about the thread-safety of `spi_transceive()`. Apparently, this particular driver does serialize thread access to the SPIM. However, it turns out that it was timing out while waiting for the transaction to complete. I happened to be using a rather slow SPI clock (125KHz) that aggravates the problem. The calculated timeout was too short and no attempt is made to abort the SPI transaction. Subsequently, after the driver returned with an the timeout error, the transaction completed (after the interrupt fired)  leaving the SPIM driver in a corrupt state.

By necessity, I'm using a Nordic SDK v1.7.1 version of Zephyr, but this issue seems to still be present in the main branch.

Here is a patch showing my solution. Comments welcome.

```
diff --git a/drivers/spi/spi_nrfx_spim.c b/drivers/spi/spi_nrfx_spim.c
index 43502a6ec7..12695e2996 100644
--- a/drivers/spi/spi_nrfx_spim.c
+++ b/drivers/spi/spi_nrfx_spim.c
@@ -239,6 +239,30 @@ static int transceive(const struct device *dev,
        transfer_next_chunk(dev);

        error = spi_context_wait_for_completion(&dev_data->ctx);
+        if( error ) {
+            /* This is typically a -ETIMEDOUT error.
+               In this case, we MUST assume that the SPIM
+               transfer is still proceeding and that the
+               completion sync semaphore is in an unknown
+               state and that an interrupt indicating the
+               end of the transfer and setting the sync
+               semaphore MAY still happen.
+             */
+           const struct spi_nrfx_config *dev_config = get_dev_config(dev);
+            /* Disable the transfer interrupt. */
+            nrf_spim_int_disable(dev_config->spim.p_reg, NRF_SPIM_INT_END_MASK);
+            /* Abort the transfer */
+            nrfx_spim_abort(&dev_config->spim);
+            /* Since the chip-select is set to false in the ISR
+               which calls transfer_next_chunk() and the ISR MAY NOT
+               have happened, we need to deassert the chip-select here.
+             */
+           spi_context_cs_control(&dev_data->ctx, false);
+            /* Ensure the sync semaphore is reset in case an
+               interrupt actually did happen.
+             */
+            k_sem_reset(&dev_data->ctx.sync);
+        }
    }

    spi_context_release(&dev_data->ctx, error);
```

mnmoron, 2022-12-17:
After a review of the other SPI drivers in the same directory that use `spi_context_wait_for_completion()` it *seems* that this may be a rather common problem if `-ETIMEOUT` is returned.

anangl, 2023-01-20:
@mnmoron Thank you for reporting this! There are actually two problems here: the first is that the expected transaction time is calculated incorrectly (and this is addressed in #53957) and the second is that transaction timeouts are not handled properly in nRF SPI drivers (this is addressed in #53965). Please check if the provided solutions work for you.

mnmoron, 2023-01-20:
@anangl Thank you for addressing this issue. I reviewed your changes and from what I can tell they accomplish approximately the same thing. However, as I mentioned in the original submission, we are using an older Nordic fork of the Zephyr code and upgrading is problematic at this point. Therefore, I won't be able to test your changes in the foreseeable future.
At this point, I am satisfied and you may close this issue.

PR #53965 (2023-02-03): drivers: spi_nrfx_spi[m]: Handle transaction timeouts properly
Add code that acts accordingly when a transaction does not complete in the expected time. It makes sure that the transaction is aborted so that no unexpected interrupt occurs afterwards and it also cleans up after that abort so that the driver can handle further requests.

Fixes #53158.
closed 2023-02-03 by mbolivar-nordic