zephyrproject-rtos/zephyr #53164: mgmt: mcumgr: NMP Timeout with smp_svr example

xudongzheng, 2022-12-18:
**Describe the bug**

Project `samples/xudongzheng/smp_svr` doesn't work with `mcumgr` when building with latest Zephyr. The same project (with minor changes) built with Zephyr 3.1 (2ddd73feafd3316af2c547c34d6969bea637d5c6) works without an issue. Board `nrf52840dongle_nrf52840` was used to reproduce this.

**To Reproduce**

Steps to reproduce the behavior:

1. Checkout attached project [smp_svr.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/10253144/smp_svr.tar.gz) at commit c19e94e227589626fc7b859f35e28c2cd4f8ce29.
2. Build project with `bash build.sh`.
3. Flash with `west flash`.
4. Run `mcumgr --conntype serial --connstring "dev=/dev/cu.usbmodem141301" echo hey`

**Expected behavior**

A `hey` response is expected.

Doing the same with Zephyr 3.1 and commit
04375ff679e1dd61dcc47faa9f934f04838e6f49 from the project works as expected.

**Impact**

This project prevents upgrade to Zephyr 3.3.

**Logs and console output**

Zephyr 3.1 (working):

```
michaelscott@mad163:~$ time mcumgr --conntype serial --connstring "dev=/dev/cu.usbmodem141301" echo hey
hey
mcumgr --conntype serial --connstring "dev=/dev/cu.usbmodem141301" echo hey  0.01s user 0.01s system 68% cpu 0.030 total
```

Zephyr 3.3 (not working):

```
$ time mcumgr --conntype serial --connstring "dev=/dev/cu.usbmodem141301" echo hey
Error: NMP timeout
mcumgr --conntype serial --connstring "dev=/dev/cu.usbmodem141301" echo hey  0.01s user 0.01s system 0% cpu 10.033 total
```

**Environment (please complete the following information):**
 - OS: macOS 12.6.1
 - Toolchain gnuarmemb
 - Commit SHA or Version used: 9cf89603dfffd320d9b3d127e5cccc1d8c40b24e

nordicjm, 2022-12-19:
Git bisect seems to show 4328f165a8ad8965ea697e71128b7a5666dd75f4 as cause
CC @de-nordic

nordicjm, 2022-12-19:
https://github.com/zephyrproject-rtos/zephyr/pull/52900 will fix this issue

PR #52900 (2022-12-19): mgmt: mcumgr: Replace non-zephyr cmake functions with zephyr versions
    This fixes an issue whilst investigating using iterable sections,
    which cannot be used when the non-zephyr prefixed functions are
    used. It also resolves a possible critical bug intoduced with the
    MCUmgr rework whereby functions or elements seem to have silently
    dropped.

Fixes #53164
closed 2022-12-19 by carlescufi