zephyrproject-rtos/zephyr #53172: SHTCx driver wrong negative temperature values

stanvn, 2022-12-18:
**Describe the bug**
There is a error in calculating negative temperature values in the SHTCx driver. The line below causes the problem. The fraction part of the temperature is calculated using `15625U`, however this converts `val->val2`  to a unsigned integer while it should be a signed integer containing a negative value.  
https://github.com/zephyrproject-rtos/zephyr/blob/9cf89603dfffd320d9b3d127e5cccc1d8c40b24e/drivers/sensor/shtcx/shtcx.c#L56 
Simply removing the `U` will fix this.

**To Reproduce**
I've placed the sensor in the freezer to reproduce the bug. 

**Expected behavior**
Expect `val->val2` to be a negative value once the temperature drops below 0.

**Impact**
Wrong readout values for temperatures below 0. 

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - Linux Ubunut (22.04)
 - Zephyr 3.2.0 (main branch)

henrikbrixandersen, 2023-06-12:
@stanvn Good catch. Are you able to submit a PR for fixing this?

stanvn, 2023-06-13:
I'll try to find some time for this, but I cannot guarantee anything.

PR #59270 (2023-06-15): drivers: sensor: shtcx: fix val2 calculation
The calculation of the sensor val2 did not correctly take the sign into account.

Fixes #53172.
closed 2023-06-15 by nashif