zephyrproject-rtos/zephyr #53173: HCI-UART: unable to preform a DFU - GATT CONN timeout

JesDebono, 2022-12-19:
**Describe the bug**
Dual processor solution where the host doesn't have a Bluetooth radio and is connected to a Bluetooth SoC via a HCI-UART interface. 
Having both applications running, able to connect and send echo commands using the Nordic 'Device Manager' app. Then trying to preform a DFU via the Nordic 'nRF Connect' app. The DFU will start hit 3% then just drops out.

 - What target platform are you using?
The host is a STM32L496 MCU running the smp_svr example https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/mgmt/mcumgr/smp_svr. 
While the controller is a Nordic nrf52832_XXAA running the HCI-UART sample application https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/hci_uart.

 - What have you tried to diagnose or workaround this issue?
 I have tried to change the buffer size allocations on both the host and controller which doesn't change the behaviour.
Sometimes the connection will just drop out with no addition log information.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect the 2 dev boards togethers with the following overlay definitions for the UART interface.

    HCI-UART
        zephyr,bt-mon-uart = &uart0;
        zephyr,bt-c2h-uart = &uart0;
        zephyr,console = &uart0;
	zephyr,shell-uart = &uart0;
	zephyr,uart-mcumgr = &uart0;

        arduino_serial: &uart0 {
	    status = "okay";
	    compatible = "nordic,nrf-uart";
	    current-speed = <115200>;
	    tx-pin = <6>;
	    rx-pin = <8>;
	    rts-pin = <5>;
	    cts-pin = <7>;
	    hw-flow-control;
       };
    
   SMP_SVR
         zephyr,bt-uart = &usart3;

       &usart3 {
	    pinctrl-0 = <&usart3_tx_pd8 &usart3_rx_pd9 &usart3_rts_pb14 &usart3_cts_pb13>;
	    pinctrl-names = "default";
	    current-speed = <115200>;
	    hw-flow-control;
	    status = "okay";
       };

2.  Using smp_svr project as the baseline for the host, build a signed image to load onto the  nucleo_l496zg board.
  Using hci_uart as the baseline for the controller build and load onto the nrf52832-mdk board.

3. Add a log entry and rebuild another version of the smp_svr application for the DFU image update. 
4. From Nordic 'nRF Connect' app Connect to the 'Zephyr' device.
5. Tap the DFU icon and select the smp_svr application firmware you just modified and rebuilt to update the host with. 
6. Select mode as 'Test and Confirm' and the press okay.
    You should see the 'nRF Connect' app bring up the transfer speed graph and after about 5 seconds just disconnects.

**Expected behavior**
A successful DFU image should be downloaded to the host MCU and rebooted into it.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Show stopper unless there is another clean way of providing Bluetooth functionality with DFU updates for non Bluetooth enabled devices.

**Logs and console output**
Logs attached.
HCI-UART
[hci_uart log.txt](https://github.com/zephyrproject-rtos/zephyr/files/10256075/hci_uart.log.txt)

SMP_SVR
[smp_svr RTT log.txt](https://github.com/zephyrproject-rtos/zephyr/files/10256077/smp_svr.RTT.log.txt)

nRF Connect
[nRF Connect log.txt](https://github.com/zephyrproject-rtos/zephyr/files/10256074/nRF.Connect.log.txt)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: gcc: 10_2020-q4-major
 - Commit SHA or Version: Zephyr v3.2.0

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nordicjm, 2022-12-19:
@Thalley can you look into this?

Thalley, 2022-12-19:
> @Thalley can you look into this?

Nope :D This is outside my scope of knowledge

nordicjm, 2022-12-19:
@jhedberg are you able to investigate this?
```
00> [00:00:30.037,000] <err> bt_driver: Unknown H:4 type 0x1b
00> [00:00:30.037,000] <err> bt_driver: Unknown H:4 type 0x00
00> [00:00:30.037,000] <err> bt_driver: Unknown H:4 type 0x08
00> [00:00:30.038,000] <err> bt_driver: Unknown H:4 type 0xfb
00> [00:00:30.038,000] <err> bt_driver: Unknown H:4 type 0x13
```

JesDebono, 2022-12-20:
@carlescufi I suspect that would be the case that you can't share console/logging with the HCI pipe. During my testing no data was sent through the console and now the console/logging is no longer on uart0 for the HCI-UART application.

In term of providing full logs of HCI traffic between ST and nRF, it's a bit hard on the ST side as enabling full logging crashes the application before making a Bluetooth connection with the 'nRF Connect' app.

[hci_uart with HCI data payload.txt](https://github.com/zephyrproject-rtos/zephyr/files/10265107/hci_uart.with.HCI.data.payload.txt)
[smp_svr with HCI payload crash.txt](https://github.com/zephyrproject-rtos/zephyr/files/10265110/smp_svr.with.HCI.payload.crash.txt)

The first set of logs above is where I added extra logging to print out the HCI traffic in the smp_svr application in the Fn: process_rx file h4.c i.e.

static inline void process_rx(void)
{
	BT_INFO("remaining %u discard %u have_hdr %u rx.buf %p len %u",
	       rx.remaining, rx.discard, rx.have_hdr, rx.buf,
	       rx.buf ? rx.buf->len : 0);

	LOG_HEXDUMP_INF(rx.buf->data, rx.buf->len, "BUF:");      /* This is causing a crash in hci_core.c ln:327 as shown in smp_svr log. */
	
	if (rx.discard) {
		rx.discard -= h4_discard(h4_dev, rx.discard);
		return;
	}
.....

While the HCI traffic for the HCI_UART application is logged at the application level in Fn: h4_send in main.c

static int h4_send(struct net_buf *buf)
{
	LOG_DBG("buf %p type %u len %u", buf, bt_buf_get_type(buf),
		    buf->len);
	
       LOG_HEXDUMP_DBG(buf->data, buf->len, "BUF:");               /* Extra log entry */
	
	net_buf_put(&uart_tx_queue, buf);
.....

Removing the problem logging statement from h4.c below another set of logs have been generated trying to preform a DFU update.

[hci_uart with HCI data payload dfu update.txt](https://github.com/zephyrproject-rtos/zephyr/files/10265180/hci_uart.with.HCI.data.payload.dfu.update.txt)
[smp_svr with HCI payload dfu update crash.txt](https://github.com/zephyrproject-rtos/zephyr/files/10265182/smp_svr.with.HCI.payload.dfu.update.crash.txt)
[nRF Connect log with HCI data payload.txt](https://github.com/zephyrproject-rtos/zephyr/files/10265183/nRF.Connect.log.with.HCI.data.payload.txt)

I hope this helps.

nordicjm, 2022-12-20:
The default settings for this are basic, i.e. DLE update but no larger than 27 byte packets, 68 byte HCI command buffer, 3 data buffers, 768 byte receive thread size. The MTU is 512 bytes, so it could be that the amount of data is too much or there are insufficient buffers to hold received packets. Try increasing all of these and seeing if the issue is resolved

JesDebono, 2022-12-23:
@nordicjm thank-you for your suggestion, modifying the buffer sizes further I have been able to do a complete download about 5% of the time. I have attached logs showing a complete download as well as a few failed downloads. 
It looks like something is causing a misalignment in the bt_recv buffer where results in an unknown H:4 type byte a number of times further resulting in a system crash.
If someone who is more familiar with the code can look at the logs and suggest further improvements that would be great. I feel we are almost there in making this feature more reliable. I have also include my configuration settings for both projects here.

Configuration

[dbg_hci_uart.txt](https://github.com/zephyrproject-rtos/zephyr/files/10291089/dbg_hci_uart.txt)
[prj_hci_uart.txt](https://github.com/zephyrproject-rtos/zephyr/files/10291090/prj_hci_uart.txt)

[dbg_smp_svr.txt](https://github.com/zephyrproject-rtos/zephyr/files/10291094/dbg_smp_svr.txt)
[prj_smp_svr.txt](https://github.com/zephyrproject-rtos/zephyr/files/10291095/prj_smp_svr.txt)

Successful download log

[BC60 HCI-UART successful update.log](https://github.com/zephyrproject-rtos/zephyr/files/10291099/BC60.HCI-UART.successful.update.log)
[BC60 smp-svr successful update.log](https://github.com/zephyrproject-rtos/zephyr/files/10291100/BC60.smp-svr.successful.update.log)

Failed download logs

[BC60 smp-svr failed 23 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291103/BC60.smp-svr.failed.23.percent.log)
[BC60 smp-svr failed 47 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291105/BC60.smp-svr.failed.47.percent.log)
[BC60 HCI-UART failed 9 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291106/BC60.HCI-UART.failed.9.percent.log)
[BC60 HCI-UART failed 23 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291107/BC60.HCI-UART.failed.23.percent.log)
[BC60 HCI-UART failed 47 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291108/BC60.HCI-UART.failed.47.percent.log)
[BC60 smp-svr failed 9 percent.log](https://github.com/zephyrproject-rtos/zephyr/files/10291109/BC60.smp-svr.failed.9.percent.log)

JesDebono, 2023-01-16:
After tweaking the parameters even further, I have a working version that is 100% successful using Nordic Device Manager in advance mode and preforming the following steps.
1. Read the image slots.
2. Perform the upload and select an image slot i.e 2 /3 in my case.
3. Read the image slots again after the download.  
4. Perform a confirmation
5. Perform a reboot.

Using NRF Connect the image swap doesn't take place. This might be something for the Nordic developers to have a look at some stage.

I have attached my conf files for this implementation and some logs from RTT both HCI-UART and SMP-SVR. Also a wireshark sniffer log as well.

Configurations
[dbg_faisal conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/10421741/dbg_faisal.conf.txt)
[prj_faisal conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/10421742/prj_faisal.conf.txt)

Logs using Device Manager
[BC60 DFU RTT smp-svr successful download.log](https://github.com/zephyrproject-rtos/zephyr/files/10421745/BC60.DFU.RTT.smp-svr.successful.download.log)
[BC60 DFU RTT hci-uart sucessful download.log](https://github.com/zephyrproject-rtos/zephyr/files/10421749/BC60.DFU.RTT.hci-uart.sucessful.download.log)
[BC60 DFU Wireshark successful download pcapng.txt](https://github.com/zephyrproject-rtos/zephyr/files/10421750/BC60.DFU.Wireshark.successful.download.pcapng.txt)

Logs using NRF Connect
[BC60 DFU RTT smp-svr successful download didn't switch image.log](https://github.com/zephyrproject-rtos/zephyr/files/10421756/BC60.DFU.RTT.smp-svr.successful.download.didn.t.switch.image.log)
[BC60 DFU RTT hci-uart successful downloaded didn't switch image.log](https://github.com/zephyrproject-rtos/zephyr/files/10421758/BC60.DFU.RTT.hci-uart.successful.downloaded.didn.t.switch.image.log)
[BC60 DFU Wireshark successful downloaded didn't switch image pcapng.txt](https://github.com/zephyrproject-rtos/zephyr/files/10421760/BC60.DFU.Wireshark.successful.downloaded.didn.t.switch.image.pcapng.txt)

carlescufi, 2023-03-06:
Closing since reporter managed to fix the issue.

closed 2023-03-06 by carlescufi