zephyrproject-rtos/zephyr #53196: Bluetooth: Host: L2CAP net buffer pool user data size ambiguity

demantmrrv, 2022-12-19:
**Describe the bug**
When using _bt_l2cap_chan_send()_, amongst other API functions using a net_buf pool, it is necessary to reserve user data memory for at least an enum, in order for the Zephyr internal house keeping using _bt_buf_set_type()_ to work. (For other API functions it is usually memory for a pointer that needs to be reserved)

If zero user data is reserved, sending L2CAP data eventually results in a null pointer dereference in _l2cap_chan_sdu_sent_, (and possibly also _bt_l2cap_send_cb_) where _user_data->cb_ is assigned. For Zephyr 3.2.99 this will be in line 1886: **bt_conn_tx_cb_t cb = data->cb;**
But already when sending the data, memory is corrupted in _conn.c/send_acl(): **bt_buf_set_type(buf, BT_BUF_ACL_OUT)**;_

If less than an enum size is reserved we still have the latter memory corruption.

In general, the documentation leaves somewhat behind describing the need for user data reservations for buffer pools for different applications.
Also, there is generally no checks to ensure that the correct amount of needed user data has been reserved before accessing or manipulating the user data memory.

**To Reproduce**
Use _bt_l2cap_chan_send()_, with a memory pool with no user data reserved on an easily debugable platform, e.g. POSIX (pseudo code):

```
NET_BUF_POOL_DEFINE(my_pool, 1, L2CAP_MAX_MTU, 0, NULL);

buf = net_buf_alloc_len(&my_pool, 4, K_NO_WAIT);
net_buf_add_mem(buf, test_buffer, 4);
bt_l2cap_chan_send(chan, buf);
```
Using this pool definition in POSIX will still cause memory corruption, but no null pointer dereference:

`NET_BUF_POOL_DEFINE(my_pool, 1, L2CAP_MAX_MTU, 1, NULL);`

**Expected behavior**
When providing a buffer pool with insufficient user data reserved we should at least assert when accessing the memory or return an error.
For instance, for the L2CAP case, calling _bt_l2cap_chan_send()_ with a net_buf with insufficient user data memory reservation could return an -ENOBUFS error or simply assert.
The same goes for _l2cap_chan_sdu_sent_, which should assert that user data is available.

In general we call _net_buf_user_data()_ from many places, expecting the memory to be available. However, no checks are implemented. For att the user data must be reserved for the _struct bt_attr_data_ pointer and so on.

**Impact**
Annoyance (at least)

**Logs and console output**
N/A

**Environment (please complete the following information):**
 OS: Linux
 Toolchain: Zephyr SDK 0.15.0
 Version 3.2.99

**Additional context**
N/A

hermabe, 2022-12-20:
Agree that the user_data size requirements is poorly communicated. We should assert if the application supplies a buffer which does not meet the requirements. The requirement is that the user data size is at least what it is for `acl_tx_pool`. https://github.com/zephyrproject-rtos/zephyr/blob/281a846060cf3d9fe92bd2f2aa1d5c050d8e0d0f/subsys/bluetooth/host/conn.c#L88-L90

This has been the size of a pointer for a long time, but I see it recently increased in https://github.com/zephyrproject-rtos/zephyr/commit/c3e5fabbf19850b10012c4450d6fdeff45714f62. This size increase could be moved to `bt_conn_tx` instead. https://github.com/zephyrproject-rtos/zephyr/blob/281a846060cf3d9fe92bd2f2aa1d5c050d8e0d0f/subsys/bluetooth/host/conn_internal.h#L139-L147

demantmrrv, 2022-12-20:
> 

Hmmm, that is also interesting. Shouldn't that have been
```
 NET_BUF_POOL_DEFINE(acl_tx_pool, CONFIG_BT_L2CAP_TX_BUF_COUNT, 
 		    BT_L2CAP_BUF_SIZE(CONFIG_BT_L2CAP_TX_MTU), 
 		    sizeof(struct tx_meta *), NULL); 
```
Instead? Because, it is a pointer to the tx_meta that is passed using the tx_data macro - similar to acl, bt_att_tx_meta_data macros and so on.

hermabe, 2022-12-20:
> Hmmm, that is also interesting. Shouldn't that have been
>
> ```
>  NET_BUF_POOL_DEFINE(acl_tx_pool, CONFIG_BT_L2CAP_TX_BUF_COUNT,
>  		    BT_L2CAP_BUF_SIZE(CONFIG_BT_L2CAP_TX_MTU),
>  		    sizeof(struct [...]

No, because a `struct tx_meta` is stored in the user data. Currently `sizeof(struct tx_meta) > sizeof(struct tx_meta *)`, so that would not work.

demantmrrv, 2022-12-20:
> 

> No, because a `struct tx_meta` is stored in the user data. Currently `sizeof(struct tx_meta) > sizeof(struct tx_meta *)`, so that would not work.

I see. It is because how the tx_data macro is used. On my POSIX BSIM they are of the same size. but it will depend of platform, alignment and optimization of course.

jori-nordic, 2024-02-09:
L2CAP CoC doesn't use user data anymore.
Concerns still valid for conn.c but we do have a check in place now: 

https://github.com/zephyrproject-rtos/zephyr/blob/40b53d5d774705d27183aaa1e42f2f322250da10/subsys/bluetooth/host/conn.c#L422-L428

closed 2024-02-09 by jori-nordic