zephyrproject-rtos/zephyr #53198: Bluetooth: Restoring security level fails and missing some notifications

ahmedmoheb-nordic, 2022-12-19:
**Application flow**

1. Central connects to 2 peripherals
2. If this is the first connection, set security level to "Level 2: Encryption and no authentication (no MITM)."
3. If this is the first connection, both central and peripherals subscribe for notifications in both directions
4. Both sides start to send notifications to connected peer(s)
5. After some time (~65 seconds) at the peripheral side, it terminates the connection and start advertising again
6. Central discovers the peripheral and connect again for another round.

**Describe the bug**
After several connection rounds
- Central receives some notifications with an incorrect connection reference (notification was sent by the one peripheral but the notification call back was called with the connection reference of the second device) https://github.com/zephyrproject-rtos/zephyr/issues/53007
- Restoring security level failed
- A disconnect event is received at the central side with reason 0x16 `BT_HCI_ERR_LOCALHOST_TERM_CONN` without any attempt to disconnect by the application (Not sure if this is an intended behavior when security level setting fails)
- Some HCI errors
- After re-connecting again, some notifications sent by the peripheral were missed at the central side

**To Reproduce**
Steps to reproduce the behavior:
1. Pull this commit 0f216a7a7dc04f23a81aa6212362c861b0572d75
2. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Logs and console output**
Please check the attached log
[log.log](https://github.com/zephyrproject-rtos/zephyr/files/10260101/log.log)

1. At line 1303, a notification was received and associated with device `DD:D1:7D:F0:5F:33` although the notification data message which contains the sender first bytes address `cnt:7 CB:04` shows the conflict
2. At line 1305, the same notification message was received again associated with the correct sender address
3. At line 1310, Security level setting failed and a following disconnect event at the central side was received
4. At line 1340, some HCI errors were caught
5. At line 1354, after restoring the connection, the peripheral started to send notification with a zero-starting counter.
But, the first message message received by the central had a counter of 2 which shows that 2 notification messages were lost

ahmedmoheb-nordic, 2022-12-20:
Applying changes in PR https://github.com/zephyrproject-rtos/zephyr/pull/52012 fixes the issues related to receiving notifications with an incorrect connection references (point 1, 2, 3 in the above comment https://github.com/zephyrproject-rtos/zephyr/issues/53198#issue-1503098862).

But, other issues were generated related to the followings
1. Restoring security level
[image]

2. Missing some notifications from the connected peripherals after establishing the connection
[image]

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".

closed 2023-05-17 by kartben