zephyrproject-rtos/zephyr #53209: LwM2M: Replace pathstrings from the APIs

jarlamsa, 2022-12-20:
**Description**
Many of the LwM2M APIs use pathstrings to denote the path to objects and resources.
This is very inefficient way as the paths are generating more and more strings to the binary.
Also the paths are usually just converted to `lwm2m_obj_path` in the beginning of a function, so there is little to no benefit of even having these strings altogether.

**Solution**
Change the APIs so that they don't anymore contain any pathstrings and all the paths should use the `lwm2m_obj_path`.
This would reduce the memory footprint of using the LwM2M -library.

**Sidenote**
This would be quite heavy refactoring to the current APIs and would require lots of changes. Currently the LwM2M library is noted to be unstable in the Zephyr and would mean that APIs can be changed without deprecation warnings.

I would like to have comments about the schedule and should we actually use deprecation warnings before changing APIs?

rlubos, 2022-12-20:
No objections from my side. Given that the LwM2M API has not been declared stable yet, I'm ok with skipping the deprecation part.

SeppoTakalo, 2022-12-20:
I also think this would be a good change.
We can replace all macros like
```
LWM2M_PATH(MY_OBJ_ID, 0, RESOURCE)
```

with the new macro that fills the object
```
LWM2M_OBJ(MY_OBJ_ID, 0, RESOURCE);
```

And it is possible to use that macro to fill the structure in place and pass as a paramter:
```
lwm2m_notify_observer_path(&LWM2M_OBJ(MY_OBJ, inst_id, RESOURCE));
```

So, this should not be too destructive change for any developers. After all, hard coded strings have been a bad example anyway.

broglep-work, 2023-01-10:
I would also love to see that change. 

Would also help address https://github.com/zephyrproject-rtos/zephyr/issues/53676

jarlamsa, 2023-01-13:
Created PR: https://github.com/zephyrproject-rtos/zephyr/pull/53771 to fix this issue.

closed 2023-01-19 by carlescufi