zephyrproject-rtos/zephyr #53227: mgmt: mcumgr: Possible instability with USB CDC data transfer

nordicjm, 2022-12-20:
**Describe the bug**
This issue is not confirmed, a cursory glance with the nrf52840 dongle seemed to show some data reliability issues with mcumgr using the echo command. This needs testing further to ensure USB CDC works as a transport reliably with both the shell and UART mcumgr transports.

nordicjm, 2023-02-09:
Definite issue with nrf52840dongle, sending `\06\09AB8CAAAVAABCAKFhZHF0ZXN0IHRoaXMgaXMgTE9OR/+C\0a` gets no response, if enter is then sent twice, a response is finally emitted back

nordicjm, 2023-02-10:
@tmon-nordic seemed to have narrowed down this issue, it related to using USB CDC for the console and `CONFIG_UART_CONSOLE_INPUT_EXPIRED` - when this is enabled and data is sent after it has elapsed, the data is essentially corrupt/incomplete, so when sending the above string, sometimes it might think tab has been pressed, or it might have received the full string except the final \n and need 1 or 2 \n characters before sending the output. Is there a way this can be fixed? Since USB CDC is not a traditional UART, does powering it down even make sense since USB is still going to be active?

tmon-nordic, 2023-02-10:
I doubt the `CONFIG_UART_CONSOLE_INPUT_EXPIRED` is really relevant to CDC ACM. It is the host that decides if USB device should be suspended or not.

de-nordic, 2023-02-10:
@nordicjm Milestone on bugfix is 3.3, do you want to change milestone here?

PR #54704 (2023-02-10): shell: backend: Set RX size default to 256 if shell MCUmgr is enabled
This fixes an issue whereby when USB CDC is used for receiving MCUmgr commands, the commands are corrupted, invalid or messed up by increasing the receive buffer size so that it can handle at least 1 full MCUmgr fragment.

Fixes #53227
closed 2023-02-10 by stephanosio