zephyrproject-rtos/zephyr #53231: drivers/flash/flash_stm32l5_u5.c : unable to use full 2MB flash

tpennors, 2022-12-20:
**Describe the bug**
With the current driver, it is not possible to access more than 1MB of the 2MB of stm32u58x MCU

**To Reproduce**
Change `storage_partition` in dts file (b_u585 board for example) to use offset after 1MB 
Build settings sample
west build -b b_u585i_iot02a samples/subsys/settings -p

**Expected behavior**
It is possible to write at this offset of flash

**Additional context**
correction seems to be to add a second bank for stm32u58x boards in `drivers/flash/flash_stm32l5_u5.c:l384`

FRASTM, 2022-12-21:
When extending the storage partition to the bank2 (> 0x810 0000) 

		storage_partition: partition@f0000 {
			label = "storage";
			reg = <0x000f0000 DT_SIZE_K(256)>;
		};

I can read/write flash there for example with samples/drivers/flash_shell : 
```
uart:~$ flash_sample write_pattern 0x120000 0x100
Reading back written bytes:
00 01 02 03 | 04 05 06 07 | 08 09 0a 0b | 0c 0d 0e 0f
10 11 12 13 | 14 15 16 17 | 18 19 1a 1b | 1c 1d 1e 1f
20 21 22 23 | 24 25 26 27 | 28 29 2a 2b | 2c 2d 2e 2f
30 31 32 33 | 34 35 36 37 | 38 39 3a 3b | 3c 3d 3e 3f
40 41 42 43 | 44 45 46 47 | 48 49 4a 4b | 4c 4d 4e 4f
50 51 52 53 | 54 55 56 57 | 58 59 5a 5b | 5c 5d 5e 5f
60 61 62 63 | 64 65 66 67 | 68 69 6a 6b | 6c 6d 6e 6f
70 71 72 73 | 74 75 76 77 | 78 79 7a 7b | 7c 7d 7e 7f
80 81 82 83 | 84 85 86 87 | 88 89 8a 8b | 8c 8d 8e 8f
90 91 92 93 | 94 95 96 97 | 98 99 9a 9b | 9c 9d 9e 9f
a0 a1 a2 a3 | a4 a5 a6 a7 | a8 a9 aa ab | ac ad ae af
b0 b1 b2 b3 | b4 b5 b6 b7 | b8 b9 ba bb | bc bd be bf
c0 c1 c2 c3 | c4 c5 c6 c7 | c8 c9 ca cb | cc cd ce cf
d0 d1 d2 d3 | d4 d5 d6 d7 | d8 d9 da db | dc dd de df
e0 e1 e2 e3 | e4 e5 e6 e7 | e8 e9 ea eb | ec ed ee ef
f0 f1 f2 f3 | f4 f5 f6 f7 | f8 f9 fa fb | fc fd fe ff
```

FRASTM, 2022-12-21:
Can you see the DBANK option bit of your device with the STM32CubeProgrammer 

[image]

tpennors, 2023-01-03:
Yes DBANK is activated as we use TF-M:
[image]

Could we deactivate this DANK bit in TF-M to access full 2MB of flash ?

erwango, 2023-01-05:
@tpennors can you provide the abstract of device tree you're using ?

tpennors, 2023-01-05:
Here is an extract of our dts :
```
/*
 * Copyright (c) 2021 STMicroelectronics
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/dts-v1/;
#include "lpp4_dl_proto0-common.dtsi"

/ {
	model = "LPP4 DL Proto-0 board based on stm32u585 MCU";
	compatible = "sofrel,lpp4_dl_proto0";

	sram3: memory@20040000 {
		reg = <0x20040000 DT_SIZE_K(524)>;
	};

	chosen {
		zephyr,console = &uart4;
		zephyr,shell-uart = &uart4;
		zephyr,sram = &sram3;
		zephyr,flash = &flash0;
		zephyr,bt-uart = &usart1;
		zephyr,code-partition = &slot1_partition;
		zephyr,entropy = &rng_psa;
	};

	rng_psa: entropy_psa_crypto {
		compatible = "zephyr,psa-crypto-rng";
		status = "okay";
		label = "psa_entropy";
	};
};

&flash0 {

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/*
		 * Following flash partition is compatible with requirements
		 * given in TFM configuration given for current board.
		 * It might require adjustment depending on evolutions on TFM.
		 */

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(208)>;
			read-only;
		};
		/* Secure image primary slot */
		slot0_partition: partition@34000 {
			label = "image-0";
			reg = <0x00034000 DT_SIZE_K(256)>;
		};
		/* Non-secure image primary slot */
		slot1_partition: partition@74000 {
			label = "image-1-nonsecure";
			reg = <0x00074000 DT_SIZE_K(1312)>;
		};
		/* NVS partition storage, for:
			- ID Card
			- littlefs sample
			- anjay_coap_client sample
		*/
		storage_partition: partition@1bc000 {
			label = "storage";
			reg = <0x001bc000 DT_SIZE_K(16)>;
		};
		/* NVS partition for temporary subsys config persistent backend */
		storage_config_nvs: partition@1c0000 {
			label = "storage_config_nvs";
			reg = <0x001c0000 DT_SIZE_K(16)>;
		};
	};
};
```

FRASTM, 2023-01-09:
With that partition, the page layout detects  256 pages of = 8192   : 2 banks are contiguous

tpennors, 2023-01-11:
In your example, do you have TF-M activated ?

FRASTM, 2023-01-13:
> In your example, do you have TF-M activated ?

No, no TFM ; that's not specified, is it.

tpennors, 2023-01-13:
Yes we use TF-M in our application.

We saw that `regression.sh` script put DBANK option byte. Is it absolutely necessary for TF-M applications ? If it is necessary, is it possible to use 2MB of flash when DBANK is activated ?

FRASTM, 2023-01-25:
Unfortunately, I am not able yet to build with TFM the b_u585i_iot02a_ns

erwango, 2023-01-27:
I'm able to reproduce the issue. Investigating

> We saw that regression.sh script put DBANK option byte. Is it absolutely necessary for TF-M applications ? If it is necessary, is it possible to use 2MB of flash when DBANK is activated ?

I guess this is optional, but has to be configured in TF-M. Please check with ST TF-M support for this.

erwango, 2023-01-31:
@tpennors There's an issue in TF-M which prevent to use some NS zones of the flash.
This could be fixed with a fix in TF-M: https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/84

Note that this fix is on TF-M v1.7.0, but should also apply to TF-M v1.6.0.

Then, I strongly encourage you, if not done already to update following TF-M file in order to configure TF-M partition to your use case: https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/master/platform/ext/target/stm/b_u585i_iot02a/include/flash_layout.h
(Note: Don't take into account the comment "Flash layout for b_u585i_iot02a  with BL2 (multiple image boot):", it's not up to date with following xxxx_SIZE definitions)

Basically, flash partition described in Zephyr only applies to non secure.
If you want to make changes to secure, this should be done in this file in TF-M.
But you should ensure both are aligned.

@microbuilder I'm not sure how to consider https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/84 (should it be merged or not in our fork). It's the real fix for an actual bug, but I'm unsure how this will be fixed in upstream.

abelradenac, 2023-02-01:
Hello, I am working on the same team and project as @tpennors

While investigating the issue, these are the test results I obtained using various configurations with the shell command _$ flash write_pattern 0x100000 0x10_ using the flash_shell sample app on a stm32u585 MCU  :

### Tests setup
1.  TF-M enabled, DBANK enabled
2.  TF-M enabled + patch erwango bd3b17b47e , DBANK enabled
3.  TF-M disabled, DBANK enabled
4.  TF-M disabled, DBANK disabled

### Results
1. CPU exception: Division by zero @ drivers/flash/flash_page_layout.c:27
2. CPU exception: Division by zero @ drivers/flash/flash_page_layout.c:27
3. CPU exception: Division by zero @ drivers/flash/flash_page_layout.c:27
4. CPU exception: Division by zero @ drivers/flash/flash_page_layout.c:27

TF-M does not seem to be the issue here, but the DBANK option byte changes the behaviour of the function _drivers/flash/flash_stm32l5_u5.c : flash_stm32_page_layout()_, still leading to the same CPU exception.

### Issues details
The implementation of the function _drivers/flash/flash_stm32l5_u5.c : flash_stm32_page_layout()_ could possibly be incriminated. Especially this part:
[image]

#### Test scenarios 1->3
DBANK is enabled so we execute the first annotated block in line 383, FLASH_PAGE_NB=128 and FLASH_PAGE_SIZE=8192 (checked with debugger), resulting in a single 1MiB bank configuration. Because the FLASH_PAGE_NB is defined as the number of pages for a single bank in _modules/hal/stm32/stm32cube/stm32u5xx/soc/stm32u585xx.h_:
[image].

#### Test scenarios 4
No flash layout is configured at all for the stm32u5 as a consequence of the macro condition directive in line 389.

### Possible fix
All the aforementioned test scenarios succeded with this fix.

When DBANK is enabled, add a configuration for the second bank that is contiguous with the first.
When DBANK is disabled, add a configuration for the stm32u5 SOC.
[image]

Here is the patch if you want to give it a try: [flash_stm32l5_u5.c.patch](https://github.com/zephyrproject-rtos/zephyr/files/10560030/flash_stm32l5_u5.c.patch)

erwango, 2023-02-02:
@abelradenac In order to ease the tracking of issues and fixes, please open a new bug description for each new point.

erwango, 2023-02-02:
@abelradenac Yes, I reproduced the issue initially described by @tpennors. The TF-M changed proposed fixed the issue on my side.
Note that I'm testing on top of main branch (which includes https://github.com/zephyrproject-rtos/zephyr/pull/51525 which fixes the issue you mention), and from what I can see you're using an older version.

abelradenac, 2023-02-02:
We are currently rebased on Zephyr 3.2.0, and when @tpennors initated the thread when we were on the 3.1.0, we failed to mention that.

The initial issue of this thread is then resolved with the PR https://github.com/zephyrproject-rtos/zephyr/pull/51525 you mentioned, this thread can be closed and we'll create a new thread for the case of DBANK deactivated.

Thanks for your time

microbuilder, 2023-02-03:
> Note that this fix is on TF-M v1.7.0, but should also apply to TF-M v1.6.0.

@erwango I don't see this fix upstreal in TF-M 1.7.0, which is also what we're currently using the the last commit of TF-M in Zephyr? https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/tree/platform/ext/target/stm/common/stm32u5xx/secure/target_cfg.c

> @microbuilder I'm not sure how to consider [zephyrproject-rtos/trusted-firmware-m#84](https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/84) (should it be merged or not in our fork). It's the real fix for an actual bug, but I'm unsure how this will be fixed in upstream.

As mentionned in the PR, this should first be fixed upstream in TF-M, then we can cherry pick that into the Zephyr fork, unless I'm missing something and this is entirely Zephyr-specific and there is a compelling reason not to fix it upstream first?

erwango, 2023-02-03:
> As mentionned in the PR, this should first be fixed upstream in TF-M, then we can cherry pick that into the Zephyr fork, unless I'm missing something and this is entirely Zephyr-specific and there is a compelling reason not to fix it upstream first?

Sure, I've pushed this PR to allow testing. I actually expects @jamike to provide a more complete fix. I'll keep you posted.

erwango, 2023-02-06:
@microbuilder I've replaced my commit with @jamike's which was merged in between (https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/19226/2/platform/ext/target/stm/common/stm32u5xx/secure/target_cfg.c)

microbuilder, 2023-02-06:
> @microbuilder I've replaced my commit with @jamike's which was merged in between (https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/19226/2/platform/ext/target/stm/common/stm32u5xx/secure/target_cfg.c)

Approved the TF-M PR, but we'll need a Zephyr PR to update the west manifest and kick off a CI run. Change looks good, though, thanks.

PR #54512 (2023-02-09): west.yml: tf-m: Fix SAU configuration on U5
Fix SAU configuration on U5 to allow non secure application to use end of flash which is not used by secure.

Fixes #53231
commit cd33ba9a (2023-02-09): west.yml: tf-m: Fix SAU configuration on U5

Fix SAU configuration on U5 to allow non secure application to use
end of flash which is not used by secure.

Fixes #53231
closed 2023-02-09 by carlescufi