zephyrproject-rtos/zephyr #53240: Task Watchdog Fallback Timeout Before Installing Timeout - STM32

bdesterBE, 2022-12-20:
**Describe the bug**
The task watchdog using a hardware watchdog fallback must be fed upon initialization, but it is not specified/unclear in the Task Watchdog documentation that this should be the case. Not doing so will result in a hardware watchdog timeout.

Digging into the source code, I have found that calling task_wdt_init(hw_wdt) when using the hardware watchdog fallback feature enabled with `CONFIG_TASK_WDT_HW_FALLBACK` will result in a hardware watchdog that must be fed, regardless of whether task_wdt_add() has been called yet. My suspicion for why this is the case, at least with the STM32 LL drivers, is that the watchdog appears to be enabled down the call stack with LL_IWDG_Enable(iwdg) from the call to wdt_install_timeout() found in task_wdt_init(). My understanding is that wdt_setup() should first be called to enable a hardware watchdog that must be fed, which does happen in task_wdt_add(). The following is where the call to LL_IWDG_Enable(iwdg) occurs:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/watchdog/wdt_iwdg_stm32.c#L139

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 I am using a custom board with an STM32 H743VIT, however a Nucleo H743ZI dev board should replicate the issue.
 - What have you tried to diagnose or workaround this issue?
 The issue is fixed by increasing the value of `CONFIG_TASK_WDT_MIN_TIMEOUT`, as specified should be done in the task watchdog source code docstring (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/task_wdt/task_wdt.c#L94). However, I still believe this is incorrect functionality, as the hardware watchdog should not need to be fed unless a task watchdog channel has been added.

**To Reproduce**
Steps to reproduce the behavior:
1. add the following configurations to the prj.conf file:
 - CONFIG_WATCHDOG=y
 - CONFIG_WDT_DISABLE_AT_BOOT=y
 - CONFIG_TASK_WDT=y
 - CONFIG_TASK_WDT_HW_FALLBACK=y
3. make a call to task_wdt_init() with the provided hardware watchdog device
4. west build
5. west flash
6. See the system restart repeatedly due to a hardware watchdog failure.

**Expected behavior**
The hardware watchdog fallback should not trigger unless a task watchdog channel has been added via the task_wdt_add() call.

**Impact**
Unclear intended functionality.

**Logs and console output**
Screenshot showing immediate reboot due to hardware watchdog immediately triggering from task watchdog initialization.
[image]

Call stack showing path for task_wdt_init().
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.10
 - Zephyr SDK 0.13.1, Zephyr 3.1.0

erwango, 2022-12-22:
@alexanderwachter Do you confirm the issue ?

alexanderwachter, 2022-12-22:
@erwango I never used the task watchdog. Maybe @martinjaeger can help?

martinjaeger, 2022-12-22:
This looks like a bug in the IWDG driver and not the task watchdog.

I think we made a mistake in #44972. The [docs](https://docs.zephyrproject.org/latest/hardware/peripherals/watchdog.html#c.wdt_setup) clearly state that the watchdog should only be activated in `wdt_setup()`, but it is enabled already in the `wdt_install_timeout()` function.

erwango, 2023-01-03:
@FRASTM, can you have a check ?

FRASTM, 2023-01-03:
Running  the samples/subsys/task_wdt on a stm32h735 disco kit 
When using the iwdg  clocked by the LSI just enabling the iwdg (and disabling the wwdg)

```
*** Booting Zephyr OS build zephyr-v3.2.0-2521-g6db3d7c96ae2 ***
Task watchdog sample application.
Main thread still alive...
Control thread started.
Main thread still alive...
Main thread still alive...
Main thread still alive...
Control thread getting stuck...
Task watchdog channel 1 callback, thread: control
Resetting device...*** Booting Zephyr OS 
```

When using the wwdg, the APB3 freq must be adjusted to match the expected WDT timeout value : 
```
&rcc {
	d1ppre = <8>;
};
```
enabling the wwdg (and disabling the iwdg)

```
*** Booting Zephyr OS build zephyr-v3.2.0-2521-g6db3d7c96ae2 ***
Task watchdog sample application.
[00:00:00.008,000] <dbg> wdt_wwdg_stm32: wwdg_stm32_install_timeout: prescaler: 128
[00:00:00.016,000] <dbg> wdt_wwdg_stm32: wwdg_stm32_install_timeout: Desired WDT: 520000 us
[00:00:00.025,000] <dbg> wdt_wwdg_stm32: wwdg_stm32_install_timeout: Set WDT:     533820 us
Main thread still alive...
Control thread started.
Main thread still alive...
Main thread still alive...
Main thread still alive...
Control thread getting stuck...
Task watchdog channel 1 callback, thread: control
Resetting device...*** Booting Zephyr OS
```

FRASTM, 2023-01-03:
also  fitting when 
```
&rcc {
	d1ppre = <16>;
};
```
and then prescaler is 64

bdesterBE, 2023-01-10:
Thank you for testing this on hardware @FRASTM. I am a little confused about what exactly the changes to the clock prescaler are doing.

Something that I did not make entirely clear in my original post is that the issue I am seeing occurs when the main thread is blocked in between calls to `task_wdt_init()` and `task_wdt_add()`. The task_wdt sample makes these calls consecutively, but the behavior can be seen if you add a sleep between them (should be ~520ms, 522 does it for me - the sample changes the `TASK_WDT_MIN_TIMEOUT` value to 500).

What I think a possible fix may be is moving the call to `wdt_install_timeout()` in task_wdt.c from `task_wdt_init()` to `task_wdt_add()`, after and only after the first task channel is added. There of course may be a more elegant solution, however.

cmorganBE, 2023-01-10:
@bdesterBE can you provide a link to the example with modifications as you've described that would show the failure?

cmorganBE, 2023-01-10:
@bdesterBE and actually, is there a test in the test suite for the task watchdog? it should be possible to add a test case that does the init call, then sleeps and then goes onto add a task, that would let us test against the behavior described in the documentation.

martinjaeger, 2023-01-11:
@cmorganBE Unfortunately, there is no test for the task watchdog in addition to the sample. I agree it would make sense to test more different use cases than just the sample. I'll add that to my ToDo list.

However, as stated above, I think this is not a bug in the task watchdog, but in the IWDG driver. The hardware watchdog should *not* yet be enabled in `wdt_install_timeout`. @FRASTM what do you think?

martinjaeger, 2023-01-11:
@bdesterBE can you try if reverting c2db84ee8af2d87db0f2b60a984a756cbbb6405b fixes your issue?

FRASTM, 2023-01-11:
Indeed, installing the watchdog should preceed starting the watchdog 
SO that In the wdt_iwdg_stm32.c driver we should  LL_IWDG_Enable in the iwdg_stm32_setup and not in the iwdg_stm32_install_timeout. The pb with this sequence is that the refMan config is not well applied (reason for PR https://github.com/zephyrproject-rtos/zephyr/pull/44972/ )

FRASTM, 2023-01-11:
The [board_porting](https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/hardware/porting/board_porting.rst) does not explicitely mention a format like "C01"

bdesterBE, 2023-01-11:
@martinjaeger just dropping [c2db84e](https://github.com/zephyrproject-rtos/zephyr/commit/c2db84ee8af2d87db0f2b60a984a756cbbb6405b) seems to cause other issues.

`*** Booting Zephyr OS build zephyr-v3.0.0-4356-gd895c97f1cdf  ***
Task watchdog sample application.
task wdt init failure: -19
[00:00:00.057,000] <err> task_wdt: hw_wdt install timeout failed: -19
Control thread started.
Control thread getting stuck...
*** Booting Zephyr OS build zephyr-v3.0.0-4356-gd895c97f1cdf  ***`

FRASTM, 2023-01-12:
> @martinjaeger just dropping [c2db84e](https://github.com/zephyrproject-rtos/zephyr/commit/c2db84ee8af2d87db0f2b60a984a756cbbb6405b) seems to cause other issues.
> 

could please specified the board under test

bdesterBE, 2023-01-12:
> > @martinjaeger just dropping [c2db84e](https://github.com/zephyrproject-rtos/zephyr/commit/c2db84ee8af2d87db0f2b60a984a756cbbb6405b) seems to cause other issues.
> 
> could please specified the board under test

Sorry, this was testing on a Nucleo H743ZI development board

FRASTM, 2023-01-13:
> > > @martinjaeger just dropping [c2db84e](https://github.com/zephyrproject-rtos/zephyr/commit/c2db84ee8af2d87db0f2b60a984a756cbbb6405b) seems to cause other issues.
> > 
> > 
> > could please specified the board under test
> 
> Sorry, this was testing on a Nucleo H743ZI development board

Could you check with  b6dba19c26fd79283f34a115dccffd663b803ec0

bdesterBE, 2023-01-13:
I can confirm that PR #53690 fixes the issue
Adding aforementioned sleep to task_wdt sample:
```
*** Booting Zephyr OS build zephyr-v3.2.0-3375-gb6dba19c26fd ***
Task watchdog sample application.
Control thread started.
Main thread still alive...
Main thread still alive...
Main thread still alive...
Control thread getting stuck...
Task watchdog channel 0 callback, thread: control
Resetting device...
```
checking out one commit before #53690 with sleep added to sample:
```
*** Booting Zephyr OS build zephyr-v3.2.0-3374-gfed4e1857f5f ***
Task watchdog sample application.
*** Booting Zephyr OS build zephyr-v3.2.0-3374-gfed4e1857f5f ***
Task watchdog sample application.
*** Booting Zephyr OS build zephyr-v3.2.0-3374-gfed4e1857f5f ***
Task watchdog sample application.
*** Booting Zephyr OS build zephyr-v3.2.0-3374-gfed4e1857f5f ***
Task watchdog sample application.
*** Booting Zephyr OS build zephyr-v3.2.0-3374-gfed4e1857f5f ***
Task watchdog sample application.
...
```

closed 2023-01-18 by carlescufi