zephyrproject-rtos/zephyr #53242: Controller in HCI UART RAW mode responds to Stop Discovery mgmt command with 0x0b status code

MrMinos, 2022-12-20:
**Describe the bug**
The controller (through HCI UART, in raw mode) responds to Stop Discovery mgmt command with `0x0b` status code.

Minimally reproducible steps:
  1. The board flashed with `v3.2.0` `samples/bluetooth/hci_uart` comes up.
  1. Launch bluetoothctl and run `scan on`.
  1. The controller receives Start Discovery mgmt command and responds with `0x00`.
  1. The controller successfully scans for the devices, we can verify via `btmon` and `dbus-monitor`
  1. Issue `scan off` command on `bluetoothctl`.
  1. The controller responds to the Stop Discovery mgmt command with `0x0b`.

Please also mention any information which could help others to understand the problem you're facing:
 - What target platform are you using?
     - `nrf52840` Makerdiary M2 Card
 - What have you tried to diagnose or workaround this issue?
     - We were able to duplicate the problem with only using `bluetoothctl`. Typical application behavior is scan on for 5 seconds and wait 5 seconds, then turn scan off and turn scan on in 5 seconds.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
    - It is not a regression, as it is also occasionally observed in `v3.1.0`. However, the controller has some success with Start and Stop Discovery commands in `v3.1.0` but eventually runs into the same issue. Bisected for this [commit](https://github.com/zephyrproject-rtos/zephyr/commit/0eec4344728bfec35a8c2b4459af2b481eed52bc) which seems to make the difference between the two versions to cause the failure to occur on the first scan rather than running the scan for about 10 to 15 minutes alternating between `scan on` and `scan off`.

**Expected behavior**
The controller should respond with 0x00 for the Stop Discovery mgmt command.

**Impact**
We cannot utilize the platform to perform BLE device scans with bluez.

**Logs and console output**
[logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/10273315/logs.zip)

**Environment (please complete the following information):**
For building firmware:
 - OS: macOS
 - Toolchain: `gnuarmemb`
 - Version: `9-2020-q2-update`

Target:
 - OS: Linux (`armv7l`)

closed 2023-01-16 by MrMinos

cvinayak, 2023-01-17:
@MrMinos Please let know the reason for closing? Is this no more reproducible?

MrMinos, 2023-01-17:
@cvinayak It is reproducible, however Zephyr was not the root cause of the problem.

broon-s, 2023-02-21:
@MrMinos we experience the same behavior, but only when there are more than ~100 devices around.
We are debugging this issue for quite some time. Could you please share what the root cause of the problem was and how you solved it?

We run a similiar system with a custom nrf52840 module connected via UART. The system is a raspberry Pi (armv7) platform with rasbian bullseye, bluez 5.55 and Zephyr LTS 2.7.X.

We created a ticket for our problem #54412 with more detailed information.

Hope to hear from your experiences...