zephyrproject-rtos/zephyr #53253: LE Audio: Increase BAP unicast BSIM test coverage

Thalley, 2022-12-21:
**Is your enhancement proposal related to a problem? Please describe.**
Currently the BAP unicast is not testing much as it previously has not been possible, but it should be now.

**Describe the solution you'd like**
Increase the BAP unicast BSIM test coverage by testing more than just codec configure, i.e. qos configure, enable, disable, update metadata etc. Everything before starting the actual stream (which needs support for connecting CIS) should be possible. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2023-01-25:
This issue is blocked by https://github.com/zephyrproject-rtos/zephyr/issues/53446

stephanosio, 2023-02-07:
Re-opening since https://github.com/zephyrproject-rtos/zephyr/issues/53253 is supposedly only a partial solution.