zephyrproject-rtos/zephyr #53254: Bluetooth: bt_conn_foreach() reports unstable conn ref before the connection is completed

ahmedmoheb-nordic, 2022-12-21:
**Describe the bug**
Calling bt_conn_foreach() in a loop to loop over active connection doesn't grantee that reported connection reference is usable.
bt_conn_foreach() reports unstable connection reference before reporting that connection reference by the 'connected' callback.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout this commit 6bc9f02d15c19fdbab54ea2fbcf88fe53553655e
2. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Expected behavior**
[image]

jhedberg, 2023-01-04:
`bt_conn_foerach()` ensures that there's a valid reference before calling the callback, and then releases the reference after the callback. So I doubt the issue is with `bt_conn_foreach()`. @ahmedmoheb-nordic could you study the code a bit further to figure out where the issue is? (it could also be in the test app)

carlescufi, 2023-02-09:
Closing as duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/53996

closed 2023-02-09 by carlescufi