zephyrproject-rtos/zephyr #53301: Bluetooth: Controller: Cannot recreate CIG

Thalley, 2022-12-22:
**Describe the bug**
It is not possible to recreate the CIG if no CIS has been connected, significantly limiting the usage of unicast ISO. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add `CONFIG_BT_CTLR_CENTRAL_ISO=y` to `tests/bluetooth/bsim_bt/bsim_test_audio/prj.conf`
2. Apply https://github.com/zephyrproject-rtos/zephyr/pull/53300 and https://github.com/zephyrproject-rtos/zephyr/pull/53291
3. Build and run the `tests/bluetooth/bsim_bt/bsim_test_audio/test_scripts/unicast_audio.sh`
4. See the following error on the second iteration (the tests runs in multiple iterations):
```
d_00: @00:00:04.330994  ASSERTION FAIL [!cig] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_central_iso.c:235
d_00: @00:00:04.330994  @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_central_iso.c:235
d_00: @00:00:04.330994  [00:00:04.330,993] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
d_00: @00:00:04.330994  [00:00:04.330,993] <err> os: z_fatal_error: Current thread: 0x81c38a0 (BT TX)
d_00: @00:00:04.330994  [00:00:04.330,993] <err> os: k_sys_fatal_error_handler: Halting system
```

**Expected behavior**
Expect to be able to create and delete (remove) a CIG using the same CIG ID multiple times. 

**Impact**
Since the ISO implementation reuses the CIG IDs, this is a major issue for ISO central. 

**Logs and console output**
See above

**Environment (please complete the following information):**
 - Commit SHA or Version used: 02b83892d2da063e665b114438b21a4f3b2443a2

**Additional context**
N/A

wopu-ot, 2023-01-03:
@mtpr-ot does this ring any bells?

mtpr-ot, 2023-01-03:
> @mtpr-ot does this ring any bells?

No, I would expect that to work.

mtpr-ot, 2023-01-03:
> > @mtpr-ot does this ring any bells?
> 
> No, I would expect that to work.

On second thought, I think there was an issue there, which Vinayak pointed out. Will take a look.

cvinayak, 2023-02-06:
@Thalley is this still an issue upstream in v3.3.0-rc1 ? if not, can we close this as fixed?

Thalley, 2023-02-06:
> @Thalley is this still an issue upstream in v3.3.0-rc1 ? if not, can we close this as fixed?

No, this was fixed by https://github.com/zephyrproject-rtos/zephyr/pull/53450 - Closing

mtpr-ot, 2023-02-06:
> > @Thalley is this still an issue upstream in v3.3.0-rc1 ? if not, can we close this as fixed?
> 
> No, this was fixed by #53450 - Closing

Hang on - my fix was removed by @cvinayak's copmmit, so the question is valid. Please check.

mtpr-ot, 2023-02-06:
Reopened until verified on RC.

Thalley, 2023-02-06:
> Reopened until verified on RC.

I can confirm that the CI BSIM tests for unicast works on the current `main`

closed 2023-02-06 by mtpr-ot