zephyrproject-rtos/zephyr #53334: Bluetooth: Peripheral disconnected with BT_HCI_ERR_LL_RESP_TIMEOUT reason

ahmedmoheb-nordic, 2022-12-25:
**Application flow**
1. There are 4 advertising peripherals which the central connects to them one after another
2. After establishing the connection and the `connected` callback is received, `bt_conn_set_security()` is scheduled to set the connection security level to `BT_SECURITY_L2` though a delayed work queue item
3. MTU is exchanged between the central and connected peripherals

**Describe the bug**
- 2 of the connected peripherals are disconnected with a disconnect reason 0x08 `BT_HCI_ERR_CONN_TIMEOUT`
- **The 4th connected peripheral is disconnected with a disconnect reason 0x22 `BT_HCI_ERR_LL_RESP_TIMEOUT`**

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout this commit 077d18460f7a5e52d0c695d8742bae6b33125744
2. Run ~/zephyrproject/zephyr/tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh

**Logs and console output**
```
d_00: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v3.2.0-2901-g630df1aaddbc ***
d_01: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v3.2.0-2825-g3a53a010f3ec ***
d_03: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v3.2.0-2825-g3a53a010f3ec ***
d_02: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v3.2.0-2825-g3a53a010f3ec ***
d_04: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v3.2.0-2825-g3a53a010f3ec ***
d_00: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
d_00: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)
d_00: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 3.2 Build 99
d_00: @00:00:00.000000  [00:00:00.000,000] <wrn> bt_id: bt_read_static_addr: No static addresses stored in controller
d_00: @00:00:00.000000  [Central] : Initializing Test
d_01: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
d_01: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)
d_03: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
d_01: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 3.2 Build 99
d_03: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)
d_03: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 3.2 Build 99
d_01: @00:00:00.000000  [00:00:00.000,000] <wrn> bt_id: bt_read_static_addr: No static addresses stored in controller
d_01: @00:00:00.000000  [Peripheral] : Initializing Test
d_04: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
d_03: @00:00:00.000000  [00:00:00.000,000] <wrn> bt_id: bt_read_static_addr: No static addresses stored in controller
d_04: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)
d_04: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 3.2 Build 99
d_03: @00:00:00.000000  [Peripheral] : Initializing Test
d_02: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
d_04: @00:00:00.000000  [00:00:00.000,000] <wrn> bt_id: bt_read_static_addr: No static addresses stored in controller
d_02: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF52x (0x0002)
d_02: @00:00:00.000000  [00:00:00.000,000] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 3.2 Build 99
d_04: @00:00:00.000000  [Peripheral] : Initializing Test
d_02: @00:00:00.000000  [00:00:00.000,000] <wrn> bt_id: bt_read_static_addr: No static addresses stored in controller
d_02: @00:00:00.000000  [Peripheral] : Initializing Test
d_00: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: Identity: DE:CD:A2:21:EE:22 (random)
d_00: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
d_00: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0xffff
d_00: @00:00:00.004568  [Central] : Bluetooth initialized
d_01: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: Identity: CB:04:A6:15:1F:50 (random)
d_01: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
d_03: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: Identity: F4:7E:23:D3:37:EA (random)
d_01: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0xffff
d_01: @00:00:00.004568  [Peripheral] : Bluetooth initialized
d_03: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
d_03: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0xffff
d_03: @00:00:00.004568  [Peripheral] : Bluetooth initialized
d_00: @00:00:00.004938  [Central] : Scanning successfully started
d_01: @00:00:00.004938  [Peripheral] : Advertising successfully started
d_03: @00:00:00.004938  [Peripheral] : Advertising successfully started
d_04: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: Identity: DD:FA:EA:DB:58:6A (random)
d_04: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
d_04: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0xffff
d_04: @00:00:00.004568  [Peripheral] : Bluetooth initialized
d_02: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: Identity: DD:D1:7D:F0:5F:33 (random)
d_02: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
d_04: @00:00:00.004938  [Peripheral] : Advertising successfully started
d_02: @00:00:00.004568  [00:00:00.004,547] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.3 (0x0c) subver 0xffff
d_02: @00:00:00.004568  [Peripheral] : Bluetooth initialized
[... 158 lines cut ...]
d_00: @00:00:12.519927  [00:00:12.519,897] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:12.568653  [Central] : Primary Service Found
d_04: @00:00:12.619624  [00:00:12.619,598] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:12.668706  [Central] : Service Characteristic Found
d_00: @00:00:12.768602  [Central] : Setting connection store index 2 peer address DD:D1:7D:F0:5F:33 (random)
d_00: @00:00:12.768602  [Central] : [SUBSCRIBED] addr DD:D1:7D:F0:5F:33 (random) conn 2
d_02: @00:00:12.866425  [Peripheral] : LE conn param updated: DE:CD:A2:21:EE:22 (random) int 0x0020 (~40 ms) lat 0 to 400
d_00: @00:00:12.866486  [Central] : LE conn param updated: DD:D1:7D:F0:5F:33 (random) int 0x0020 (~40 ms) lat 0 to 400
d_02: @00:00:13.632531  [Peripheral] : [NOTIFICATION] addr DE:CD:A2:21:EE:22 (random) conn 0x80d3060 data cnt:0 length 220 cnt 0
d_00: @00:00:13.832689  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 2 data cnt:0 DD:D1 length 220 cnt 0
d_02: @00:00:23.633455  [Peripheral] : [NOTIFICATION] addr DE:CD:A2:21:EE:22 (random) conn 0x80d3060 data cnt:100 length 220 cnt 100
d_00: @00:00:23.829617  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 2 data cnt:100 DD:D1 length 220 cnt 100
d_02: @00:00:33.593455  [Peripheral] : [NOTIFICATION] addr DE:CD:A2:21:EE:22 (random) conn 0x80d3060 data cnt:200 length 220 cnt 200
d_00: @00:00:33.791154  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 2 data cnt:200 DD:D1 length 220 cnt 200
d_02: @00:00:43.593455  [Peripheral] : [NOTIFICATION] addr DE:CD:A2:21:EE:22 (random) conn 0x80d3060 data cnt:300 length 220 cnt 300
d_00: @00:00:43.789618  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 2 data cnt:300 DD:D1 length 220 cnt 300
d_00: @00:00:52.419784  [Central] disconnected:503 : Connection (0) @ 0x80d35c0 with Peer 7A:4D:09:81:30:B1 (random) terminated (reason 0x22)
[... cut ...]
d_02: @00:01:03.553455  [Peripheral] : [NOTIFICATION] addr DE:CD:A2:21:EE:22 (random) conn 0x80d3060 data cnt:500 length 220 cnt 500
d_00: @00:01:03.749617  [Central] : [NOTIFICATION] addr DD:D1:7D:F0:5F:33 (random) conn 2 data cnt:500 DD:D1 length 220 cnt 500
d_00: @00:01:12.388571  [Central] disconnected:503 : Connection (2) @ 0x80d3730 with Peer DD:D1:7D:F0:5F:33 (random) terminated (reason 0x15)
d_02: @00:01:12.388572  [Peripheral] disconnected:88 : Connection @ 0x80d3060 with Peer DE:CD:A2:21:EE:22 (random) terminated (reason 0x16)
```

carlescufi, 2023-03-06:
cc @jori-nordic perhaps you can show @kruithofa how to run the test.

kruithofa, 2023-03-14:
This issue does not exist anymore when rebasing on the main branch.
Closing the issue

closed 2023-03-14 by kruithofa