zephyrproject-rtos/zephyr #53335: Undeclared constants in devicetree_generated.h

knicklicht, 2022-12-25:
**Describe the bug**
When I try to build the example app `samples/hello_world` for the `litex_vexriscv` board in a clean zephyr project with:
`cat overlay.config | xargs west build -b litex_vexriscv samples/hello_world/ -- -DDTC_OVERLAY_FILE=$PWD/overlay.dts`

 I get the following error message:

`[114/126] Building C object zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/litex_timer.c.obj
FAILED: zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/litex_timer.c.obj 
ccache /home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DKERNEL -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/user/zephyrproject/zephyr/include -I/home/user/zephyrproject/zephyr/build/zephyr/include/generated -I/home/user/zephyrproject/zephyr/soc/riscv/litex-vexriscv -isystem /home/user/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include -isystem /home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/include-fixed -fno-strict-aliasing -Os -imacros /home/user/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mabi=ilp32 -march=rv32ima_zicsr_zifencei --sysroot=/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/riscv64-zephyr-elf -imacros /home/user/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/user/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/litex_timer.c.obj -MF zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/litex_timer.c.obj.d -o zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/litex_timer.c.obj -c /home/user/zephyrproject/zephyr/drivers/timer/litex_timer.c
In file included from /home/user/zephyrproject/zephyr/soc/riscv/litex-vexriscv/soc.h:11,
                 from /home/user/zephyrproject/zephyr/include/zephyr/arch/riscv/irq.h:26,
                 from /home/user/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:20,
                 from /home/user/zephyrproject/zephyr/include/zephyr/arch/cpu.h:25,
                 from /home/user/zephyrproject/zephyr/include/zephyr/kernel_includes.h:33,
                 from /home/user/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from /home/user/zephyrproject/zephyr/drivers/timer/litex_timer.c:9:
/home/user/zephyrproject/zephyr/drivers/timer/litex_timer.c: In function 'sys_clock_cycle_get_32':
/home/user/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h:4266:34: error: 'DT_N_S_soc_S_timer_e0002800_REG_NAME_uptime_latch_VAL_ADDRESS' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_timer_e0002800_REG_NAME_update_value_VAL_ADDRESS'?
 4266 | #define DT_N_INST_0_litex_timer0 DT_N_S_soc_S_timer_e0002800
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3905:33: note: in definition of macro 'DT_CAT4'
 3905 | #define DT_CAT4(a1, a2, a3, a4) a1 ## a2 ## a3 ## a4
      |                                 ^~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3501:9: note: in expansion of macro 'DT_REG_ADDR_BY_NAME'
 3501 |         DT_REG_ADDR_BY_NAME(DT_DRV_INST(inst), name)
      |         ^~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_N_INST_0_litex_timer0'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/home/user/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:326:31: note: in expansion of macro 'UTIL_CAT'
  326 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3048:27: note: in expansion of macro 'DT_INST'
 3048 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3501:29: note: in expansion of macro 'DT_DRV_INST'
 3501 |         DT_REG_ADDR_BY_NAME(DT_DRV_INST(inst), name)
      |                             ^~~~~~~~~~~
/home/user/zephyrproject/zephyr/drivers/timer/litex_timer.c:24:41: note: in expansion of macro 'DT_INST_REG_ADDR_BY_NAME'
   24 | #define TIMER_UPTIME_LATCH_ADDR         DT_INST_REG_ADDR_BY_NAME(0, uptime_latch)
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/drivers/timer/litex_timer.c:52:42: note: in expansion of macro 'TIMER_UPTIME_LATCH_ADDR'
   52 |         litex_write8(TIMER_UPTIME_LATCH, TIMER_UPTIME_LATCH_ADDR);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h:4266:34: note: each undeclared identifier is reported only once for each function it appears in
 4266 | #define DT_N_INST_0_litex_timer0 DT_N_S_soc_S_timer_e0002800
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3905:33: note: in definition of macro 'DT_CAT4'
 3905 | #define DT_CAT4(a1, a2, a3, a4) a1 ## a2 ## a3 ## a4
      |                                 ^~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3501:9: note: in expansion of macro 'DT_REG_ADDR_BY_NAME'
 3501 |         DT_REG_ADDR_BY_NAME(DT_DRV_INST(inst), name)
      |         ^~~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_N_INST_0_litex_timer0'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/home/user/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:326:31: note: in expansion of macro 'UTIL_CAT'
  326 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3048:27: note: in expansion of macro 'DT_INST'
 3048 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/devicetree.h:3501:29: note: in expansion of macro 'DT_DRV_INST'
 3501 |         DT_REG_ADDR_BY_NAME(DT_DRV_INST(inst), name)
[... 167 lines cut ...]
    status = "disabled";
};
&i2s_rx {
    status = "disabled";
};
&i2s_tx {
    status = "disabled";
};
&clock0 {
    status = "disabled";
};
&ram0 {
    reg = <0x40000000 0x800000>;
};
&dna0 {
    status = "disabled";
};
`

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 20.04)
 - Toolchain: Zephyr SDK, Litex, Xilinx ISE

**Additional context**
None

gmarull, 2022-12-28:
Please provide a branch where this can be reproduced. Looks like the overlay is also setting too many options, e.g. CONFIG_UART_LITEUART should never be selected, only CONFIG_SERIAL=y

knicklicht, 2022-12-28:
> Please provide a branch where this can be reproduced. Looks like the overlay is also setting too many options, e.g. CONFIG_UART_LITEUART should never be selected, only CONFIG_SERIAL=y

I used the version from the getting started guide. Should I clone the main branch and rebuild Zephyr?

gmarull, 2022-12-28:
> > Please provide a branch where this can be reproduced. Looks like the overlay is also setting too many options, e.g. CONFIG_UART_LITEUART should never be selected, only CONFIG_SERIAL=y
> 
> I used the version from the getting started guide which is version 0.15.2 (release). Should I clone the main branch and rebuild Zephyr?

description mentions overlays for example. Are they upstream?

knicklicht, 2022-12-28:
> description mentions overlays for example. Are they upstream?

The overlays were generated using a tool from the Litex project (https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts_zephyr.py). It generates the overlays automatically from the generated vexriscv that was created. I am unsure if the issue is due to incorrectly generated overlays or if there is a bug in Zephyr. I also mentioned the issue in their Discord but am still waiting for a response.

knicklicht, 2023-01-03:
I tried to build with `-DCONFIG_LITEX_TIMER=n` to disable the timer. In this case I can successfully build but then I get an error during linking: `undefined reference to `sys_clock_cycle_get_32'` (see full message below). This is the same function that previously produced the error. Is this expected behavior because the timer is necessary?

```
[156/166] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/user/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/user/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/timing/libsubsys__timing.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0"  -L/home/user/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mabi=ilp32  -march=rv32ima_zicsr_zifencei  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/user/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(kernel_service.c.obj): in function `cmd_kernel_threads':
/home/user/zephyrproject/zephyr/subsys/shell/modules/kernel_service.c:162: undefined reference to `sys_clock_elapsed'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(kernel_service.c.obj): in function `arch_k_cycle_get_32':
/home/user/zephyrproject/zephyr/subsys/shell/modules/kernel_service.c:232: undefined reference to `sys_clock_cycle_get_32'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/arch/common/libarch__common.a(timing.c.obj): in function `arch_k_cycle_get_32':
/home/user/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:259: undefined reference to `sys_clock_cycle_get_32'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/kernel/libkernel.a(sched.c.obj): in function `z_reset_time_slice':
/home/user/zephyrproject/zephyr/kernel/sched.c:435: undefined reference to `sys_clock_elapsed'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/kernel/libkernel.a(timeout.c.obj): in function `elapsed':
/home/user/zephyrproject/zephyr/kernel/timeout.c:64: undefined reference to `sys_clock_elapsed'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/kernel/libkernel.a(timeout.c.obj): in function `arch_k_cycle_get_32':
/home/user/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:259: undefined reference to `sys_clock_cycle_get_32'
/home/user/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/kernel/libkernel.a(timeout.c.obj): in function `z_impl_k_busy_wait':
/home/user/zephyrproject/zephyr/kernel/timeout.c:333: undefined reference to `sys_clock_cycle_get_32'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyrproject/zephyr/build
```

knicklicht, 2023-01-06:
The issue was resolved by building the litex vexriscv soc with `--timer-uptime` to enable the missing latch.

closed 2023-01-06 by knicklicht

cfriedt, 2023-01-06:
Thanks @knicklicht! I had not really had time to get to repro this yet, but I'm happy that you documented the problem and the solution 😄