zephyrproject-rtos/zephyr #53361: Adding I2C devices SX1509B to Devicetree with the same address on different busses generates FATAL ERROR.

kabachokserhii, 2022-12-28:
**Describe the bug**
Two I2C devices were connected to the device tree:
- I2C1: SX1509B Device address 0x3E
- I2C3: SX1509B Device address 0x3E

The SX1509B uses a driver from the Zephyr repository.
When building the project, we get a critical error.
We used the board supported by Zephyr nucleo_f413zh and our own added stm32f401rb, the error is the same.

**To Reproduce**
Add two devices to the device tree as shown below. Devices on different buses have the same address.

```
&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb6 &i2c1_sda_pb7>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	sx1509b_2: sx1509b@3e {
		compatible = "semtech,sx1509b";
		reg = <0x3e>;
		status = "okay";
		gpio-controller;
		#gpio-cells = <2>;
		ngpios = <16>;
		init-out-high = <0xffff>;
	};
};

&i2c3 {
	pinctrl-0 = <&i2c3_scl_pa8 &i2c3_sda_pc9>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	sx1509b_1: sx1509b@3e {
		compatible = "semtech,sx1509b";
		reg = <0x3e>;
		status = "okay";
		gpio-controller;
		#gpio-cells = <2>;
		ngpios = <16>;  
		init-out-high = <0xffff>;
	};
};
```

Also, the definition of devices in the application is given below:

```
const struct device *sx1509b_dev_1 = DEVICE_DT_GET(DT_NODELABEL(sx1509b_1));
const struct device *sx1509b_dev_2 = DEVICE_DT_GET(DT_NODELABEL(sx1509b_2)); 
```

**Logs and console output**
The output of the errors we get:
```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj
-- Found Python3: /home/sintez/zephyrproject/.venv/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/sintez/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/sintez/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: stm32f401_mc
-- Found host-tools: zephyr 0.15.0 (/home/sintez/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/home/sintez/zephyr-sdk-0.15.0)
-- Found Dtc: /home/sintez/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/sintez/zephyrproject/zephyr/boards/arm/stm32f401_mc/stm32f401_mc.dts
-- Found devicetree overlay: boards/stm32f401rb_mc.overlay
-- Generated zephyr.dts: /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/dts.cmake

warning: GPIO_SX1509B (defined at drivers/gpio/Kconfig.sx1509b:6) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: DT_HAS_SEMTECH_SX1509B_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_GPIO_SX1509B and/or look up GPIO_SX1509B in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Parsing /home/sintez/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/sintez/zephyrproject/zephyr/boards/arm/stm32f401_mc/stm32f401_mc_defconfig'
Merged configuration '/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/prj.conf'
Configuration saved to '/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/.config'
Kconfig header saved to '/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sintez/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build
-- west build: building application
[1/169] Preparing syscall dependency handling

[3/169] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/home/sintez/zephyrproject/zephyr), build: zephyr-v3.1.0-4680-g88ae9cbb9d38
[159/169] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/zephyr_pre0.map 
: && ccache /home/sintez/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/sintez/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7e-m/nofp"  -L/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build/zephyr && /usr/bin/cmake -E echo
/home/sintez/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(test.c.obj):/home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/src/test.c:42: undefined reference to `__device_dts_ord_33'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/sintez/zephyrproject/workspace/stm32f401rb_mc_pj/build
```

**Environment (please complete the following information):**

- OS: Ubuntu 22.04.1 LTS
- Toolchain: Zephyr SDK / GCC
- Zephyr version: 3.2.0

gmarull, 2022-12-28:
Please share `prj.conf` as well, it looks like a configuration issue.

```
warning: GPIO_SX1509B (defined at drivers/gpio/Kconfig.sx1509b:6) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: DT_HAS_SEMTECH_SX1509B_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_GPIO_SX1509B and/or look up GPIO_SX1509B in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
```

kabachokserhii, 2022-12-28:
> Поділіться `prj.conf`також, схоже, проблема конфігурації.
> 
> ```
> warning: GPIO_SX1509B (defined at drivers/gpio/Kconfig.sx1509b:6) was assigned the value 'y' but got
> the value 'n'. Check these unsatisfied dependencies: DT_HAS_SEMTECH_SX1509B_ENABLED (=n). See
> http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_GPIO_SX1509B and/or look up GPIO_SX1509B in
> the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
> Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
> ```

The prj.conf contains:
```
CONFIG_PRINTK=y
CONFIG_HEAP_MEM_POOL_SIZE=1024
CONFIG_GPIO=y
CONFIG_GPIO_SX1509B=y
CONFIG_I2C=y
CONFIG_ADC=y
```

gmarull, 2022-12-28:
`CONFIG_GPIO_SX1509B=y` should not be set, it is automatically selected if CONFIG_GPIO=y and device enabled in DT.

kabachokserhii, 2022-12-28:
> `CONFIG_GPIO_SX1509B=y` should not be set, it is automatically selected if CONFIG_GPIO=y and device enabled in DT.

I removed this line. It didn't fix the problem.

gmarull, 2022-12-28:
Cannot be reproduced with the provided details:

```
diff --git a/boards/arm/nucleo_f413zh/nucleo_f413zh.dts b/boards/arm/nucleo_f413zh/nucleo_f413zh.dts
index 4ce1740cb7..ad8f619059 100644
--- a/boards/arm/nucleo_f413zh/nucleo_f413zh.dts
+++ b/boards/arm/nucleo_f413zh/nucleo_f413zh.dts
@@ -98,6 +98,33 @@
 	pinctrl-names = "default";
 	status = "okay";
 	clock-frequency = <I2C_BITRATE_FAST>;
+
+	sx1509b_2: sx1509b@3e {
+		compatible = "semtech,sx1509b";
+		reg = <0x3e>;
+		status = "okay";
+		gpio-controller;
+		#gpio-cells = <2>;
+		ngpios = <16>;
+		init-out-high = <0xffff>;
+	};
+};
+
+&i2c3 {
+	pinctrl-0 = <&i2c3_scl_pa8 &i2c3_sda_pc9>;
+	pinctrl-names = "default";
+	status = "okay";
+	clock-frequency = <I2C_BITRATE_FAST>;
+
+	sx1509b_1: sx1509b@3e {
+		compatible = "semtech,sx1509b";
+		reg = <0x3e>;
+		status = "okay";
+		gpio-controller;
+		#gpio-cells = <2>;
+		ngpios = <16>;
+		init-out-high = <0xffff>;
+	};
 };
 
 &spi1 {
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 9a90c6a6a0..d0ca199ba6 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -5,6 +5,10 @@
  */
 
 #include <zephyr/kernel.h>
+#include <zephyr/device.h>
+
+const struct device *sx1509b_dev_1 = DEVICE_DT_GET(DT_NODELABEL(sx1509b_1));
+const struct device *sx1509b_dev_2 = DEVICE_DT_GET(DT_NODELABEL(sx1509b_2));
 
 void main(void)
 {
```

```
west build -b nucleo_f413zh samples/hello_world -p -- -DCONFIG_GPIO=y -DCONFIG_I2C=y -DCONFIG_ADC=y
```

closed 2022-12-28 by carlescufi

kabachokserhii, 2022-12-29:
Thanks for the help. It works.

kabachokserhii, 2022-12-29:
> edit: found the problem, driver did not support more than 1 instance. Fixed here #53373

Thanks for the help. It works.

PR #53373 (2022-12-28): drivers: gpio: sx1509b: add multi-instance support
The driver only supported one instance. Update it to support multiple instances.

Fixes #53361