zephyrproject-rtos/zephyr #53375: net: lwm2m: write method when floating point

RomainPelletant, 2022-12-28:
**Describe the bug**
When server send write operation (float) on 3340/0/5521, internal server error (500) is returned.

Please also mention any information which could help others to understand
the problem you're facing:
- SENML CBOR is used. cbor_decode_lwm2m_senml function doesn't return ZCBOR_SUCCESS and then return EBADMSG
```
dret = cbor_decode_lwm2m_senml(ICTX_BUF_R_PTR(&msg->in), ICTX_BUF_R_LEFT_SZ(&msg->in),
					   &fd->dcd, &decoded_sz);

if (dret != ZCBOR_SUCCESS) {
	ret = -EBADMSG;
	goto error;
}
```
- CBOR : it returns net_lwm2m_cbor: unable to decode a floating-point value and returns internal server error (500)

I am using leshan server with SENML CBOR (and CBOR for test).
Wireshark trace of write request 
```
0000   04 91 62 87 26 2f d4 54 8b 8c 1b a8 08 00 45 00   ..b.&/.T......E.
0010   00 4b ae 89 00 00 80 11 00 00 c0 a8 00 35 c0 a8   .K...........5..
0020   00 3c 16 35 a1 84 00 37 82 0a 48 03 ee 0b b0 2b   .<.5...7..H....+
0030   28 45 c6 6a bc c3 b4 33 33 34 30 01 30 04 35 35   (E.j...3340.0.55
0040   32 31 11 70 ff 81 a2 02 f9 4f 80 21 6c 2f 33 33   21.p.....O.!l/33
0050   34 30 2f 30 2f 35 35 32 31                                   40/0/5521
```

**To Reproduce**
Steps to reproduce the behavior:
1. lwm2m client sample
2. add res instance 5521 with post_write_callback
3. make
4. Launch write request on 3340/0/5521

**Expected behavior**
Write request deserialization of float success and then call post_write callback directly

carlescufi, 2023-01-04:
@oyvindronningstad FYI

SeppoTakalo, 2023-01-04:
I can verify that this is true. Easy to replicate by running Leshan locally and LwM2M client build on `native_posix`

rlubos, 2023-01-04:
I think that's the culprit:
[image]
Leshan appears to be using 16-bit float representation, which seemingly is not supported by zcbor. @oyvindronningstad Any insight on when can we expect zcbor to support 16-bit floats in Zephyr? I see the feature is on a PR already (https://github.com/NordicSemiconductor/zcbor/pull/285).

SeppoTakalo, 2023-01-04:
Yes. I just found the same.
https://github.com/zephyrproject-rtos/zcbor/blob/main/src/zcbor_decode.c#L762..L773

Zcbor only accepts `float` and `double`.

When I write big numbers in SenML-CBOR using Leshan, it seem to work.
So this is not LwM2M specific, should be fixed in Zcbor library.

SeppoTakalo, 2023-01-04:
Verified that PR https://github.com/NordicSemiconductor/zcbor/pull/285 fixes the issue.
If I checkout the PR only my local environment and rebuild, it seem to work OK.

RomainPelletant, 2023-01-04:
Thanks for your help

oyvindronningstad, 2023-01-04:
@rlubos I'm not sure exactly when it will come to Zephyr. I haven't decided the feature set for the next release, but when I release something, I will bring it to Zephyr as well.

broglep-work, 2023-01-10:
I only have created the downstream zcbor issue https://github.com/NordicSemiconductor/zcbor/issues/282 but not reported it here in zephyr, sorry for the extra work & investigation caused. 

We would also look forward to have that fix brought to zephyr soon. 

But we can't prioritise it on our side at the moment as it is rather low prio and we have a workaround in place

RomainPelletant, 2023-02-02:
@carlescufi Can I close this issue please?

rlubos, 2023-02-02:
@RomainPelletant Why do you want do close it? AFAIK zcbor hasn't been updated yet.

RomainPelletant, 2023-02-02:
The float PR has been merged but maybe not in zephyrproject-rtos/zcbor yet

SeppoTakalo, 2023-04-18:
This will be fixed in https://github.com/zephyrproject-rtos/zephyr/pull/56471

PR #56471 (2023-04-18): west.yml: Update zcbor to point to 0.7.0
Among other things, this update brings:

* C++ improvements
* float16 support
* Improved docs
* -Wall and -Wconversion compliance

Fixes #53375
closed 2023-04-18 by carlescufi