zephyrproject-rtos/zephyr #53379: IP address configuration for multiple interfaces

krish2718, 2022-12-28:
We have Kconfig options to choose either a static IP (`CONFIG_NET_CONFIG_MY_IPV4_ADDR`) or DHCP client or both for a single interface, but in case we have multiple interfaces all these will only be applicable to a single interface either default interface or the one passed to `net_config_init_app`.

We need to define custom Kconfig and call net APIs to configure static IP address, so, I would like to propose below enhancements:

### Requirements

- [ ] Extend IP address Kconfig options for all interfaces, possibly using interface indexing (IP_ADDR_0, IP_ADDR_1 etc)
- [ ] Add support to run DHCPv4 client per-interface, DHCPV4_IF_0
- [ ] Add ability to choose the interface order within the same type (a bit difficult, but I am thinking of device name used to register)
- [ ] Add `net shell` commands for the same to configure either from Kconfig or shell.

e.g.,

- For netusb + wifi, FIRST_IF = eth_netusb, IP_ADDR_0="10.0.0.1", DHCPV4_IF_1
- For thread + wifi, FIRST_IF=wlan0, DHCPV4_IF_0, IP_ADDR_1="20.0.02"

Of course the above Kconfig suggestions are just to explain the requirement, feel free to come up with a better scheme.

@rlubos @rado17 @sachinthegreen

jukkar, 2022-12-29:
Something like this have already been discussed in #29750, perhaps we could continue the design ideas in that issue.
Anyway, I think we got into conclusion there that DT could be used for the configuration instead of creating new Kconfig options.

krish2718, 2022-12-29:
Thanks, I only did a preliminary search for existing issues, it satisfies my requirements, though using DTS for SW sounds a bit weird, but I do agree with the rationale.

carlescufi, 2023-06-21:
Closing this in favor of #29750, since this is almost identical.

closed 2023-06-21 by carlescufi