zephyrproject-rtos/zephyr #53430: drivers: display: otm8009a: import 3rd-party source

dariis, 2023-01-03:
## Origin

STM32Cube_FW_H7_V1.9.1/Drivers/BSP/Components/otm8009a/
or https://github.com/STMicroelectronics/stm32-otm8009a

## Purpose

Initializes the OTM8009A display over MIPI DSI.

## Mode of integration

Integrate to the main tree as a driver.

## Maintainership

Should be maintainable by the Zephyr display driver (or `mipi_dsi`) maintainer.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/53044

Discussion: https://github.com/zephyrproject-rtos/zephyr/discussions/52860#discussioncomment-4502450

## Description

Basically only used the MIPI DSI init sequence from the original source.

Driver is necessary for the ST B-LCD40-DSI1 shield.

## Dependencies

Zephyr MIPI DSI API.

## Revision

not applicable.

## License

Verbatim import as `BSD-3-Clause` license and then changed to `Apache-2.0`

gmarull, 2023-01-03:
I think it should just be implemented as a native display driver, using the MIPI-DSI API. What is the value of adding more vendor modules that bring poor quality code, not aligned with upstream?

FYI, this is an outdated (and probably needs fixes) native implementation: https://github.com/teslabs/zephyr/commits/feature/stm32-dsi.

dariis, 2023-01-09:
As proposed by @gmarull, let's do a native implementation.

closed 2023-01-09 by dariis