zephyrproject-rtos/zephyr #53437: XNUCLEO-F411RE: Error: ** Unable to reset target **

s163555, 2023-01-03:
**Describe the bug**
Similar to #50590  I am experiencing an issue with resetting the STM32F411RE with west and openocd. I am using the board XNUCLEO-F411RE (an improved NUCLEO-F411RE board from WaveShare). The bug is easily reproduced by flashing zephyr unto the board once, and then after this, the board is unable to reset. 
If the on-board reset button is physically pressed and released with the right timing, the flashing will succeed. If not, it will give the error message:
Error: timed out while waiting for target halted
embedded:startup.tcl:1070: Error: ** Unable to reset target **

Please also mention any information which could help others to understand
the problem you're facing:
 - nucleo_f411re (xnucleo_f411re)
 - I have tried two programmers STLINK V2mini and STLINK V3set, both work can reset with STM32CubeProgrammer 
 - Is this a regression? To my knowledge this has been an issue since I started trying to flash to the board

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Impact**
An annoyance, but a significant annoyance since every flashing requires multiple times if I cannot time the button release at the right time.

**Logs and console output**
zephyr SDK 15.2
```
C:\Users\Jeppe\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: C:/Users/Jeppe/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.12.0-rc3 (2022-12-20-12:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J40S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.289186
Info : [stm32f4x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f4x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       running

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Error: timed out while waiting for target halted
TARGET: stm32f4x.cpu - Not halted
FATAL ERROR: command exited with status 1: 'C:/Program Files/OpenOCD/bin/openocd.exe' -s 'C:/Users/Jeppe/zephyrproject/zephyr/boards/arm/nucleo_f411re\support' -s C:/Users/Jeppe/zephyr-sdk-0.15.2/sysroots/AMD64-pokysdk-linux/usr/share/openocd/scripts -f 'C:/Users/Jeppe/zephyrproject/zephyr/boards/arm/nucleo_f411re\support\openocd.cfg' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase C:/Users/Jeppe/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset init' -c 'verify_image C:/Users/Jeppe/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

platformIO (zephyr v. 3.2.0)
```
Processing nucleo_f411re (platform: ststm32; framework: zephyr; board: nucleo_f411re)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html
PLATFORM: ST STM32 (15.4.1) > ST Nucleo F411RE
HARDWARE: STM32F411RET6 100MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-zephyr @ 2.20500.210226+sha.9f9f3e2f
 - tool-cmake @ 3.21.3
 - tool-dfuutil @ 1.11.0
 - tool-dtc @ 1.4.7
 - tool-ninja @ 1.9.0
 - tool-openocd @ 2.1100.211028 (11.0)
 - tool-stm32duino @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\nucleo_f411re\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.4% (used 4437 bytes from 131072 bytes)
Flash: [          ]   2.6% (used 13636 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\nucleo_f411re\firmware.elf
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Error: timed out while waiting for target halted
embedded:startup.tcl:1070: Error: ** Unable to reset target **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1107
at file "embedded:startup.tcl", line 1070
*** [upload] Error 1
```

**Environment (please complete the following information):**
 - Windows
 - Zephyr SDK and platformIO (zephyr v3.2.0)
 - git describe: zephyr-v3.2.0-3013-ge4fcb32451

**Additional context**

henrikbrixandersen, 2023-01-03:
Can you reproduce this on current Zephyr `main` without using PlatformIO?

s163555, 2023-01-03:
> Can you reproduce this on current Zephyr `main` without using PlatformIO?

I have included the zephyr sdk in the post, and it is the same behavior

erwango, 2023-01-03:
> I have both the NUCLEO-F411RE board and the XNUCLEO-F411RE board, and it seems that the regular NUCLEO-F411RE can flash, so it might be specific to the WaveShare board

Ok, I just found out that XNUCLEO-F411RE is actually not the usual, ST, NUCLEO-F411RE.
I'm converting this to a discussion as this is not impacting an upstream board.